raspberrypi/tools #29: Add Python support to arm-linux-gnueabihf-gdb from linaro toolchain

sschuberth, 2015-02-09:
Please re-build arm-linux-gnueabihf-gdb from the linaro toolchain with Python scripting support so that e.g. Qt Creator (version 3.1.1 in my case) can be used for debugging.

Ruffio, 2015-06-27:
@sschuberth is this still an issue?

sschuberth, 2015-06-29:
Well, AFAICT these binaries haven't changed since 2014, so yes, it's still an issue. Also see e.g. [this blog post](https://jeanleflambeur.wordpress.com/2014/09/28/remote-debugging-on-the-raspberry-pi-from-qtcreator/) which suggests to use Ubuntu's `gdb-multiarch` package as a work-around instead.

ths80, 2015-12-03:
Any news on this Issue? I've got the same problem.

Marox44, 2016-04-24:
Today I encountered the same issue. Currect version of Qt Creator requires gdb built with python support, and using standard MinGW windows gdb doesn't work obviously.

It would be really usefull to have **arm-linux-gnueabihf-gdb ** with python support, so it would be possible to cross-compile using toolchain and then remotely debug.

Regards.

megakoresh, 2017-07-31:
This is still a problem, hello? Is this repo abandoned?

JamesH65, 2017-08-01:
No, not abandoned AFAIK. Very little churn though. 

Not sure what our position is on rebuilding the toolchain.