raspberrypi/tools #31: mkknlimg produces numerous 'Use of uninitialized value' warnings

srcshelton, 2015-03-10:
By default, mkknlimg doesn't "`use warnings;`".  I was trying to debug what turned out to be a permissions issue, but I noticed that with this added in, the result I get is:

```
$ sudo mkknlimg --dtok arch/arm/boot/zImage /boot/0/kernel/kernel-3.18.9 
Use of uninitialized value $pos in substitution (s///) at /usr/bin/mkknlimg line 229.
Use of uninitialized value $pos in concatenation (.) or string at /usr/bin/mkknlimg line 230.
tail: +: invalid number of bytes
DT: y
Use of uninitialized value in pack at /usr/bin/mkknlimg line 247.
Use of uninitialized value in pack at /usr/bin/mkknlimg line 247.
Use of uninitialized value in pack at /usr/bin/mkknlimg line 247.
Use of uninitialized value in pack at /usr/bin/mkknlimg line 247.
```

... although it does appear to have worked:

```
$ knlinfo /boot/0/kernel/kernel-3.18.9
Kernel trailer found at 1973120/0x1e1b80:
  KVer: ""
  DTOK: true
```

I'm not sure whether the kernel version not being populated is significant.

srcshelton, 2015-03-10:
So, the issues seem to be:
* Not all versions of 'unxz' support '`--single-stream`' as a parameter - release 5.0.5 doesn't, for example;
* If decompression fails, '`push @atoms, [ 'KVer', $res->{''} ];`' appears to be using `$res` undefined;
* In the above example, `$pos` is determined to be 15161, but the following command fails:
```
$ tail -c+15161 "arch/arm/boot/zImage" | unxz >/dev/null ; echo $?
unxz: (stdin): Compressed data is corrupt
1
```
... although 5349640 bytes of (good?) data is output before this abort occurs.

pelwell, 2015-03-11:
It's on the list of things to look at. If you have a proposed solution to the unxz problem (this was a contributed format, and not one we use) then you can outline it here or submit a Pull Request.

srcshelton, 2015-03-15:
It seems that Ubuntu (and so, presumably, Debian and Raspian) ship with xz-utils 5.1.1alpha.  I'm building on Gentoo where the stable version of xz-utils is 5.0.8, and this lacks the '--single-stream' option.  However, given that the kernel is being built on the same host, I'd be surprised if the xz-compressed image is something that the system xz binary can't deal with.  However, to confirm I'll upgrade to release 5.2.1 and see if anything changes.
My guess is that there's some un- (or, at least, differently-) compressed blob being appended to the raw kernel zImage, and unxz is aborting when it hits this...

Ruffio, 2015-06-27:
@srcshelton is this still an issue?

srcshelton, 2015-09-10:
Sorry for the delay - it's looking much better now:

```
$ rpi-mkknlimg --dtok arch/arm/boot/zImage test.img
DT: y
DDT: n
283x: n
$ rpi-knlinfo test.img
Kernel trailer found at 1986128/0x1e4e50:
  KVer: "?"
  DTOK: y
  DDTK: n
  283x: n
```

... so it's not picking up the kernel version correctly, but I'm not sure whether that's merely a cosmetic problem:

```diff
$ diff -u <( xxd arch/arm/boot/zImage ) <( xxd test )
--- /dev/fd/63
+++ /dev/fd/62
@@ -124131,3 +124131,8 @@
 001e4e20: 504e 1e00 144e 1e00 6852 1e00 ac36 0000  PN...N..hR...6..
 001e4e30: 6452 1e00 644e 1e00 544e 1e00 0c09 0000  dR..dN..TN......
 001e4e40: 5c4e 1e00 604e 1e00 584e 1e00 0000 0000  \N..`N..XN......
+001e4e50: 0000 0000 0000 0000 0000 0000 0400 0000  ................
+001e4e60: 3238 3378 0000 0000 0400 0000 4444 544b  283x........DDTK
+001e4e70: 0100 0000 0400 0000 4454 4f4b 3f00 0000  ........DTOK?...
+001e4e80: 0100 0000 4b56 6572 4400 0000 0400 0000  ....KVerD.......
+001e4e90: 5250 544c                                RPTL
```

... there does appear to be a `KVer` value in there, so this may be a problem?

For reference, the stock (v7) kernel image ends:

```
003d8450: 382e 3320 3230 3134 3033 3033 2028 7072  8.3 20140303 (pr
003d8460: 6572 656c 6561 7365 2920 2863 726f 7373  erelease) (cross
003d8470: 746f 6f6c 2d4e 4720 6c69 6e61 726f 2d31  tool-NG linaro-1
003d8480: 2e31 332e 312b 627a 7232 3635 3020 2d20  .13.1+bzr2650 -
003d8490: 4c69 6e61 726f 2047 4343 2032 3031 342e  Linaro GCC 2014.
003d84a0: 3033 2920 2920 2338 3130 2053 4d50 2050  03) ) #810 SMP P
003d84b0: 5245 454d 5054 2054 7565 2041 7567 2031  REEMPT Tue Aug 1
003d84c0: 3820 3135 3a33 323a 3132 2042 5354 2032  8 15:32:12 BST 2
003d84d0: 3031 3500 bf00 0000 4b56 6572 0001 0000  015.....KVer....
003d84e0: 0400 0000 5250 544c                      ....RPTL
```

Is there an documentation as to what the `--ddtk` and `--283x` options do (the latter marking the kernel as RPi2(-only))?

P.S. The usage string needs an update ;)

pelwell, 2015-09-10:
This version is now deprecated. It has been replaced by the kernel-specific version found in scripts/ within the kernel tree. The --ddtk option may make an appearance in future kernel releases (Hint: DDTK stands for Dynamic Device Tree Kernel, but don't enable this unless you want to stop the firmware from loading your overlays).

The knlinfo output is reporting that it has found a KVer string of "?", which indicates that mkknlimg has failed to locate the kernel version string in your build. Note that the version string matching feature is primitive (it searches for a string beginning "Linux version " followed by a non-zero digit), so if you have changed this is in some way then it won't work. Fortunately nothing currently makes use of this information, but it is sometimes handy to be able to interrogate an image with knlinfo.

The 283x flag indicates that the kernel has been built for an upstream target (using bcm2835_defconfig), and causes the firmware to load bcm2835-*.dtb (and, in the future, bcm2836-*.dtb) instead of the usual bcm270x versions.

popcornmix, 2016-02-14:
Closing as this repo's tool is no longer used.

closed 2016-02-14 by popcornmix