raspberrypi/tools #34: c++11 compiler specifications

groolot, 2015-04-03:
It seems that the compiler does not implement ``c++11`` standard language as ``-std=c++11`` but only with ``-std=c++0x`` with the defined macro ``__cplusplus != 201103L``.

How can we use the ``c++11`` standard with this compiler ?

Annonomus-Penguin, 2015-06-03:
If I recall correctly, older versions of GCC require `c++0x` whereas newer ones use `c++11x`. Thus, I think it is the version of GCC that determines the type you use.

popcornmix, 2015-06-03:
Are you talking about the gcc cross compiler in this repo?
``echo "int x;" > x.cpp && arm-linux-gnueabihf-gcc -c -std=c++11 x.cpp``
runs without complaint for me.

Ruffio, 2015-06-27:
@groolot any update/clarification on this issue?

popcornmix, 2016-02-14:
I've added linaro gcc 4.9.3 to tools repo.
I suspect this can be closed.

closed 2016-02-14 by popcornmix