raspberrypi/tools #37: Python 2 configure error: "You must get working getaddrinfo() function."

josephbisch, 2015-06-23:
I am trying to compile Python 2 using the tools in `arm-bcm2708/gcc-linaro-arm-linux-gnueabihf-raspbian-x64/bin`. During the configure process, I get the following output:

```
checking for getaddrinfo... yes
checking getaddrinfo bug... yes
Fatal: You must get working getaddrinfo() function.
       or you can specify "--disable-ipv6".
```

This seems to be a bug with these tools, because the same Python compiles for x86_64 Linux fine.

I know the configure script is picking up the tools correctly, because I see output like `checking host system type... arm-unknown-linux-gnueabihf` and `checking for arm-linux-gnueabihf-gcc... arm-linux-gnueabihf-gcc`.

I am using the following Makefile to build Python. It is part of the depends system adopted from Bitcoin Core designed to build software dependencies in a reproducible way. It doesn't work standalone, but you can still get an idea of what it does.

```
package=python2
$(package)_version=2.7.9
$(package)_download_path=https://www.python.org/ftp/python/$($(package)_version)
$(package)_file_name=Python-$($(package)_version).tgz
$(package)_sha256_hash=c8bba33e66ac3201dabdc556f0ea7cfe6ac11946ec32d357c4c6f9b018c12c5b

define $(package)_config_cmds
  $($(package)_autoconf) --build=$(BUILD)
endef

define $(package)_build_cmds
  $(MAKE)
endef

define $(package)_stage_cmds
  $(MAKE) DESTDIR=$($(package)_staging_dir) install
endef
```

As you can see it downloads Python 2.7.9 and verifies the hash. It runs `./configure` with some standard flags like `--host`. I also supply `--build`. Then it runs make and make DESTDIR=... install. This same exact Makefile works with x86_64-unknown-linux-gnu as the `--host` parameter.

The relevant section of configure.ac from Python is line [3514 of configure.ac](https://github.com/python/cpython/blob/9e14ea6a4ba2b2ec5024ba2e8be4a9550ef817b4/configure.ac#L3514). That is a link to Python 3, but you get the idea. It looks like there are a number of things checked in that one getaddrinfo() check, so I am not sure which one is actually the issue.

popcornmix, 2015-06-24:
It is much more likely an issue with the configure script rather than the toolchain.
We build a lot of code with this toolchain and it was built for us by Linaro.

I suggest you try adding ``--disable-ipv6`` to the configure line as the error suggests.

josephbisch, 2015-06-24:
Thanks for getting back.

I looked at the Python configure script closer and saw that explicitly setting `--enable-ipv6` seems to override the result of the test. The build seems to proceed fine, but I haven't actually tried out the result yet. You are probably right, but I'll leave this open until I can verify the build works.

Using `--disable-ipv6` is not an option for my case.

VijayaRanis01, 2023-10-13:
`python3-devel` installation fixed the issue

closed 2023-10-13 by popcornmix