raspberrypi/tools #38: Error with current toolchain (outdated, maybe?)

KouTatsumiya, 2015-08-03:
I've been trying to compile the RPi's kernel using Ubuntu to cross-compile. I've been using a default configuration, and every time I try and compile the kernel, I get this error:

#error Your compiler is too buggy; it is known to miscompile kernels
  ^
arch/arm/kernel/asm-offsets.c:54:2: error: #error and result in filesystem corruption and oopses.
 #error and result in filesystem corruption and oopses.

I've followed the documentation on the official Raspberry Pi website, and as someone who's new to kernel compilation, I'm not quite sure how to fix this. Maybe it's an issue that only I'm experiencing, and if so, does anyone know how to resolve this?

popcornmix, 2015-08-03:
The official kernel builds are built with this:
https://github.com/raspberrypi/tools/tree/master/arm-bcm2708/gcc-linaro-arm-linux-gnueabihf-raspbian-x64/bin

What are you using to compile?

KouTatsumiya, 2015-08-03:
The documentation told me just to download the whole tools file....if that's the case, should I just download the x64 file only?

popcornmix, 2015-08-04:
You download the whole tools repo.
What matters is what you set CROSS_COMPILE to.

KouTatsumiya, 2015-08-04:
I set it as arm-linux-gnueabihf.....is that wrong?

popcornmix, 2015-08-05:
That is correct. What does:
``arm-linux-gnueabihf-gcc --version``
report?