raspberrypi/tools #41: Toolchain to use on jessie

carlonluca, 2015-10-11:
Is the linaro toolchain currently in the repo still OK to crossbuild for Raspbian Jessie? I read it should be, but when trying to build something that links to icu I get this:

```
ICU auto-detection... ()
/opt/rpi/gcc-linaro-arm-linux-gnueabihf-raspbian/bin/arm-linux-gnueabihf-g++ -c -pipe -march=armv7-a -marm -mthumb-interwork -mfpu=neon-vfpv4 -mtune=cortex-a7 -mabi=aapcs-linux -mfloat-abi=hard --sysroot=/opt/rpi/sysroot -O2 -Wall -W -fPIC  -I. -I/opt/rpi/sysroot/home/pi/qtdeps/include -I../../../mkspecs/devices/linux-rasp-pi2-g++ -o icu.o icu.cpp
/opt/rpi/gcc-linaro-arm-linux-gnueabihf-raspbian/bin/arm-linux-gnueabihf-g++ -Wl,-rpath-link,/opt/rpi/sysroot/opt/vc/lib -Wl,-rpath-link,/opt/rpi/sysroot/home/pi/qtdeps/lib -mfloat-abi=hard --sysroot=/opt/rpi/sysroot -Wl,-O1 -o icu icu.o   --sysroot=/opt/rpi/sysroot -L/opt/rpi/sysroot/home/pi/qtdeps/lib -licui18n -licuuc -licudata
/opt/rpi/sysroot/home/pi/qtdeps/lib/libicui18n.so: undefined reference to `__cxa_throw_bad_array_new_length@CXXABI_1.3.8'
collect2: error: ld returned 1 exit status
Makefile:110: recipe for target 'icu' failed
make: *** [icu] Error 1
ICU disabled.
```

I suppose that symbol was added in 4.9 (I may be wrong). Also I see that the compiler in the Jessie image is 4.9.2. So is this toolchain OK for Jessie? Or should I use another one?

popcornmix, 2015-10-13:
We still build kernels and userland libs (/opt/vc/lib) with the [linaro](https://github.com/raspberrypi/tools/tree/master/arm-bcm2708/gcc-linaro-arm-linux-gnueabihf-raspbian-x64) toolchain which works fine with jessie.
Did you build libicui18n.so yourself, or was that built with a newer compiler?

carlonluca, 2015-10-13:
I'm sorry, the path was misleading. The error:

    undefined reference to `__cxa_throw_bad_array_new_length@CXXABI_1.3.8'

is returned when building against the icu library provided with jessie. I also tried to rebuild icu myself using the linaro toolchain from this repo, and my builds work fine then, but at the end, for some reason unknown to me, my applications won't start. The dynamic linker seems to somehow ignore my icu lib, although it is in the search path (http://pastebin.com/nMNZn3gs, http://pastebin.com/9a28k4Sx). I tried with ubuntu crosscompiler but I suppose it expects a different sysroot layout. Do you have any other advice?

carlonluca, 2015-10-13:
By reading the elf of the icu I built I found out it was built softfp. Building it properly solved. So it was simply my fault. Not sure why the dynamic linker wasn't saying anything more useful.
The first part regarding the toolchain remains, but actually icu is the only one that showed problems so far.

teamonhand, 2016-02-09:
Dear Dev team,
Are there any plans to upgrade the compilers (arm-bcm2708 folder) to sync up with 4.9.2 (used in Jessie). I would like to make modules using the same version of gcc as the jessie distribution and that would really help. 
Thanks,
Jack

popcornmix, 2016-02-10:
@teamonhand are you happy with the linked toolchain?
We could add it to repo if it is useful.

teamonhand, 2016-02-10:
It did the job for me. I think it would be convenient to have it included in the repo.

popcornmix, 2016-02-14:
I've added linaro gcc 4.9.3 to tools repo.
@carlonluca does that work for you?

carlonluca, 2016-02-14:
Awesome, theoretically it should yes. I don't know when I'll have the time to rebuild all my projects to test so if you want you can close the issue and I'll reopen in case of problems.

closed 2016-02-14 by popcornmix