raspberrypi/tools #44: Usb boot could be updated.

ED6E0F17, 2016-01-01:
It would be nice to be able to boot a model A from usb, but there may be issues with usbboot/buildroot.elf expecting 512Mb of ram.

The supplied usbboot/buildroot.elf is over 12 months old, and should be updated more frequently to support the latest hardware and kernel features.

VC_BUILD_ID_TIME: Oct 28 2014

popcornmix, 2016-01-01:
ping @ghollingworth

ghollingworth, 2016-01-02:
I'm in the process of creating a new filesystem boot mechanism which means that we'll not require buildroot.elf and instead use the standard start.elf

Watch this space...

Gordon

AndyPi, 2016-02-03:
Hi, 
I have been trying to follow instructions but they dont work.  Aim is to boot the Pi0 from USB and then use ethernet over usb connectivity.  Any news or ETA.  

After hacking the patch to get the USB bits into buildroot config the compile fails with 
> of_find_device_by_node" [drivers/misc/bcm2835_smi.ko] undefined!
Im thinking the linux config is out of date. 

regards...

ghollingworth, 2016-02-03:
Not being able to build buildroot isn't a part of this, I am not going to keep the patch up to date for all buildroot versions.  I'd suggest understanding the changes I had to make to buildroot to get it to build and then implementing those on a recent version.  The linux config will be a year or more out of date now...

Gordon

ED6E0F17, 2016-02-04:
"Someone" should update the buildroot patch to copy across the device tree files - but it is quite hard to test before the firmware is updated.

It would be nice to net-boot Raspbian over USB without an sdcard.

Gadgetoid, 2016-02-05:
@AndyPi I've not been able to boot anything on the Pi0 over USB, I believe `usbbootcode.bin` simply doesn't support it at the moment. I know the tool works because I've tried it with a raw, VideoCore "blink" binary compiled with vasm, but `usbbootcode.bin` simply doesn't cause the re-enumeration and serial port increment that the tool expects.

I'm eagerly awaiting an updated version.

Apparently this already works for running custom binaries ( for baremetal tinkering ) on A+/CM, but not on the Zero: https://github.com/dwelch67/raspberrypi/issues/15

AndyPi, 2016-02-07:
Hi 
thanks for the reply, maybe I'm not going mad !  I too believe that the failure to re-enumerate to serial=1 causes the (potentially) infinite wait.

ED6E0F17, 2016-03-11:
Any news on Pi Zero support? Should the CM3 update work if I add device tree files to my "buildroot" ?

Gadgetoid, 2016-03-19:
The new CM3 update certainly causes the Pi Zero to re-enumerate to serial 1 and allow code to be uploaded, however I've got no further with getting code to run. `rpiboot` concats the FAT image I'm generating onto the end of `buildroot.elf` before uploading, so I'm inclined to think that the out-of-date `buildroot.elf` is now the culprit, but I'm honestly somewhat out of my depth here and clutching at straws!

AndyPi, 2016-05-16:
@ghollingworth 
>
I'm in the process of creating a new filesystem boot mechanism which means that we'll not require buildroot.elf and instead use the standard start.elf
Watch this space...
> 

Is there any news on new boot mechanism which avoids buildroot.  It would be very useful.  

regards
Andy

closed 2016-05-18 by ED6E0F17

lurch, 2017-01-17:
> I am not going to keep the patch up to date for all buildroot versions

That's fair enough, but it might be useful if the docs gave a buildroot git commit-hash for when your buildroot patch _was_ working :-)

ghollingworth, 2017-01-17:
I'm going to update this system completely in the next few days (all being well) which means this'll all get a lot easier, meaning no buildroot patch required anymore!