raspberrypi/tools #45: Is this cross compiler valid for Raspberry Pi 2?

vanfanel, 2016-01-14:
Hi,

The CPU core on the Pi2 is a different family from the one in the original Raspberry Pi (2708 vs 2709) and the cross compiler in this repository seems to be tailored towards the Pi 1 CPU:

```
manuel@vader:~$ arm-bcm2708hardfp-linux-gnueabi-gcc -v -Q ppp.c 
Using built-in specs.
COLLECT_GCC=arm-bcm2708hardfp-linux-gnueabi-gcc
COLLECT_LTO_WRAPPER=/home/manuel/tools/arm-bcm2708/arm-bcm2708hardfp-linux-gnueabi/bin/../libexec/gcc/arm-bcm2708hardfp-linux-gnueabi/4.7.1/lto-wrapper
Target: arm-bcm2708hardfp-linux-gnueabi
Configured with: /home/extra/crosstool/staginghf/.build/src/gcc-linaro-4.7-2012.04/configure --build=i686-build_pc-linux-gnu --host=i686-build_pc-linux-gnu --target=arm-bcm2708hardfp-linux-gnueabi --prefix=/home/dc4/tools/arm-bcm2708/arm-bcm2708hardfp-linux-gnueabi --with-sysroot=/home/dc4/tools/arm-bcm2708/arm-bcm2708hardfp-linux-gnueabi/arm-bcm2708hardfp-linux-gnueabi/sysroot --enable-languages=c,c++ --with-cpu=arm1176jzf-s --with-tune=arm1176jzf-s --with-float=hard --with-pkgversion='crosstool-NG 1.15.2' --enable-__cxa_atexit --disable-libmudflap --disable-libgomp --disable-libssp --with-gmp=/home/extra/crosstool/staginghf/.build/arm-bcm2708hardfp-linux-gnueabi/buildtools --with-mpfr=/home/extra/crosstool/staginghf/.build/arm-bcm2708hardfp-linux-gnueabi/buildtools --with-mpc=/home/extra/crosstool/staginghf/.build/arm-bcm2708hardfp-linux-gnueabi/buildtools --with-ppl=/home/extra/crosstool/staginghf/.build/arm-bcm2708hardfp-linux-gnueabi/buildtools --with-cloog=/home/extra/crosstool/staginghf/.build/arm-bcm2708hardfp-linux-gnueabi/buildtools --with-libelf=/home/extra/crosstool/staginghf/.build/arm-bcm2708hardfp-linux-gnueabi/buildtools --with-host-libstdcxx='-static-libgcc -Wl,-Bstatic,-lstdc++,-Bdynamic -lm -L/home/extra/crosstool/staginghf/.build/arm-bcm2708hardfp-linux-gnueabi/buildtools/lib -lpwl' --enable-threads=posix --enable-target-optspace --disable-nls --disable-multilib --with-local-prefix=/home/dc4/tools/arm-bcm2708/arm-bcm2708hardfp-linux-gnueabi/arm-bcm2708hardfp-linux-gnueabi/sysroot --enable-c99 --enable-long-long --with-float=hard
Thread model: posix
gcc version 4.7.1 20120402 (prerelease) (crosstool-NG 1.15.2) 
COLLECT_GCC_OPTIONS='-v' '-Q' '-mcpu=arm1176jzf-s' '-mfloat-abi=hard' '-mtls-dialect=gnu'
 /home/manuel/tools/arm-bcm2708/arm-bcm2708hardfp-linux-gnueabi/bin/../libexec/gcc/arm-bcm2708hardfp-linux-gnueabi/4.7.1/cc1 -v -iprefix /home/manuel/tools/arm-bcm2708/arm-bcm2708hardfp-linux-gnueabi/bin/../lib/gcc/arm-bcm2708hardfp-linux-gnueabi/4.7.1/ -isysroot /home/manuel/tools/arm-bcm2708/arm-bcm2708hardfp-linux-gnueabi/bin/../arm-bcm2708hardfp-linux-gnueabi/sysroot ppp.c -dumpbase ppp.c -mcpu=arm1176jzf-s -mfloat-abi=hard -mtls-dialect=gnu -auxbase ppp -version -o /tmp/cc0ugD9R.s
```
So you can see it's using Pi1 cpu compilation options/flags by default, uses the Pi1 FPU, etc...

Is there (or makes it sense) a Pi2 pre-built cross-compiler? This one has been SO useful to me thee years on the Pi1 for distcc... And building my own is normally a painful process prone to failures with crosstool-ng.

popcornmix, 2016-01-14:
By default it produces code that runs on both pi1 and pi2.
However nothing to stop you passing appropriate flags to produce a pi2 specific build flags (e.g. ``-mcpu=cortex-a7 -mfloat-abi=hard -mfpu=neon-vfpv4``) which may be more optimal.

vanfanel, 2016-01-14:
Will it also receive these flags if I use distcc on both the Pi and the compilation server to build?

popcornmix, 2016-01-14:
That would be a question for distcc. I'm sure it's possible to pass arguments, but that's not related to this cross compiler.

vanfanel, 2016-01-14:
Yes, you're right on that. Sorry, and thanks :)

closed 2016-01-14 by vanfanel