raspberrypi/tools #46: Cross-compiling on Ubuntu 14.0.4.3

solderspot, 2016-02-04:
I am trying to cross-compile a Raspberry Pi application using this tool chain on a Ubuntu 14.0.4.3 64 bit server instance but am experiencing an issue I'm not sure how to handle.

On the Ubuntu instance I have copied over the **/usr** tree from my pi to a **piroot** folder. In the CMakeLists.txt file I have:

`include_directories(SYSTEM ${PIROOT}/usr/include ...`

where **PIROOT** is the path to the **piroot** folder.

However, I get the following compiler error:

```
[ 20%] Building CXX object CMakeFiles/PiCamCVTest.dir/main.cpp.o
In file included from /home/solderspot/pidev/piroot/opt/vc/include/interface/vcos/pthreads/vcos_platform.h:59:0,
                 from /home/solderspot/pidev/piroot/opt/vc/include/interface/vcos/vcos.h:116,
                 from /home/solderspot/pidev/piroot/opt/vc/include/interface/vmcs_host/vc_dispmanx.h:33,
                 from /home/solderspot/pidev/piroot/opt/vc/include/bcm_host.h:50,
                 from /home/solderspot/pidev/camtest/mmalincludes.h:9,
                 from /home/solderspot/pidev/camtest/camera.h:3,
                 from /home/solderspot/pidev/camtest/main.cpp:5:
/home/solderspot/pidev/piroot/usr/include/stdlib.h:955:31: fatal error: bits/stdlib-float.h: No such file or directory
 #include <bits/stdlib-float.h>
                               ^
compilation terminated.
make[2]: *** [CMakeFiles/PiCamCVTest.dir/main.cpp.o] Error 1
make[1]: *** [CMakeFiles/PiCamCVTest.dir/all] Error 2

```

So the compiler is correctly referencing the **piroot/usr/include** tree but **stdlib.h**'s reference to **bits/stdlib-float.h** is failing to find it in **piroot/usr/include/arm-linux-gnueabihf**

My question is: Why does the compiler not know to search the platform specific folders? Is there some setting I'm missing?

---
Here is more info which might be helpful:

The application compiles and runs on the native system. 

I've been able to compile and run a simple hello world program with this toolchain.

This is the **CMAKE_TOOLCHAIN_FILE**:

```
 SET(CMAKE_SYSTEM_NAME Linux)
 SET(CMAKE_SYSTEM_VERSION 1)
 
 SET(TOOLROOT ${PITOOLS}/arm-bcm2708/gcc-linaro-arm-linux-gnueabihf-raspbian-x64)
 
 # specify the cross compiler
 SET(CMAKE_C_COMPILER   ${TOOLROOT}/bin/arm-linux-gnueabihf-gcc)
 SET(CMAKE_CXX_COMPILER ${TOOLROOT}/bin/arm-linux-gnueabihf-g++)
 
 # search for programs in the build host directories
 SET(CMAKE_FIND_ROOT_PATH_MODE_PROGRAM NEVER)
 # for libraries and headers in the target directories
 SET(CMAKE_FIND_ROOT_PATH_MODE_LIBRARY ONLY)
 SET(CMAKE_FIND_ROOT_PATH_MODE_INCLUDE ONLY)
 
```

Where **PITOOLS** is the path to the this repo.

And the **CMakeLists.txt** for the project is:

```
cmake_minimum_required(VERSION 2.8)
project( PiCamCVTest )
SET(COMPILE_DEFINITIONS -Werror)

include_directories(SYSTEM ${PIROOT}/usr/include ${PIROOT}/opt/vc/include ${PIROOT}/opt/vc/include/interface/vcos/pthreads ${PIROOT}/opt/vc/include/interface/vmcs_host/linux )
link_directories( ${PIROOT}/opt/vc/lib )
add_executable(PiCamCVTest main.cpp camera.cpp cameracontrol.cpp graphics.cpp)

target_link_libraries(PiCamCVTest libmmal_core.so libmmal_util.so libmmal_vc_client.so libvcos.so librt.so libbcm_host.so GLESv2 EGL libopencv_core.so libopencv_imgproc.so)

```

I've detailed my efforts in two journal posts: http://wp.me/p493sy-xY and http://wp.me/p493sy-xY

ED6E0F17, 2016-02-05:
Almost certainly not an issue with the toolchain. You have three branches on Github, and none of them have:

    include_directories(SYSTEM .... 

My raspberry has two copies of stdlib.h, but neither of them have "#include <bits/stdlib-float.h>"; the one on my X86 machine does.

ED6E0F17, 2016-02-05:
Looking at the difference between x86 jessie, and raspi wheezy, there is no simlink from /usr/include/bits to /usr/include/arm-linux-gnueabihf/bits , so you will need to add arm-linux-gnueabihf to your include path. (Assuming you have "bits/stdlib-float.h" in "/usr/include/arm-linux-gnueabihf")

solderspot, 2016-02-05:
Yes I'm using the "Raspbian GNU/Linux 8.0 (jessie)" distro and it has the different setup.

I've added the following to the toolchain file:

```
SET(FLAGS "-isystem ${PIROOT}/usr/include/arm-linux-gnueabihf")

UNSET(CMAKE_C_FLAGS CACHE)
UNSET(CMAKE_CXX_FLAGS CACHE)
SET(CMAKE_CXX_FLAGS ${FLAGS} CACHE STRING "" FORCE)
SET(CMAKE_C_FLAGS ${FLAGS} CACHE STRING "" FORCE)
```

And the files compile correctly now.

The linker is still looking in **/usr/lib..**. (i.e. from the root of the host system).

I tried adding **--sysroot=${PIROOT}** to the flags but it had no effect. I did verify the flag was getting set:

```
make -f CMakeFiles/PiCamCVTest.dir/build.make CMakeFiles/PiCamCVTest.dir/build
/usr/local/bin/cmake -E cmake_echo_color --switch= --green --progress-dir=/home/solderspot/pidev/camtest/build/CMakeFiles --progress-num=1 "Building CXX object CMakeFiles/PiCamCVTest.dir/main.cpp.o"
/home/solderspot/pidev/pitools/arm-bcm2708/gcc-linaro-arm-linux-gnueabihf-raspbian-x64/bin/arm-linux-gnueabihf-g++    -isystem /opt/vc/include -isystem /opt/vc/include/interface/vcos/pthreads -isystem /opt/vc/include/interface/vmcs_host/linux  --sysroot=/home/solderspot/pidev/piroot/ -isystem /usr/include/arm-linux-gnueabihf  -L/lib/arm-linux-gnueabihf -L/usr/lib/arm-linux-gnueabihf   -o CMakeFiles/PiCamCVTest.dir/main.cpp.o -c /home/solderspot/pidev/camtest/main.cpp
```

closed 2016-02-08 by solderspot