raspberrypi/tools #47: rpiboot problem

AndyPi, 2016-02-07:
I dont think I'm being an idiot but I can't definitely rule it out.  I'm trying to use the rpiboot program to boot Pi zero **without** SD card, which I think should be straightforward since few problems reported here or elsewhere.  But I just can't get it to work at all. I hope someone can help.

I have a Pi Zero with empty card slot.  A micro A / USB A cable connected to Pi Zero micro USB.  Host is Intel based PC with 64 bit Ubuntu 14.04 (3.19.0-33-generic).

On the host PC I have done the following (as per instructions)

```
git clone --depth=1 https://github.com/raspberrypi/tools
cd tools/usbboot
sudo apt-get install libusb-1.0-0-dev
make
	cc -g -o rpiboot main.c -lusb-1.0

sudo make install
	cp rpiboot /usr/bin
	mkdir -p /usr/share/rpiboot
	cp usbbootcode.bin /usr/share/rpiboot	
	cp msd.elf /usr/share/rpiboot
	cp buildroot.elf /usr/share/rpiboot

```
Plug the USB A (connected to Pi Zero) into my Host PC

```
dmesg
	usb 2-1.8: new full-speed USB device number 7 using ehci-pci
	usb 2-1.8: New USB device found, idVendor=0a5c, idProduct=2763
	usb 2-1.8: New USB device strings: Mfr=1, Product=2, SerialNumber=0
	usb 2-1.8: Product: BCM2708 Boot
	usb 2-1.8: Manufacturer: Broadcom
```

```
lsusb
	Bus 002 Device 007: ID 0a5c:2763 Broadcom Corp.
``` 

```
sudo rpiboot -v

	Waiting for BCM2835 ...
	Found serial = 0: writing file ./usbbootcode.bin
	libusb_bulk_transfer returned 0
	Writing 16674 bytes
	libusb_bulk_transfer returned 0
	Successful
	Waiting for BCM2835 ...
```

It just waits(and waits and waits...), and lsusb entry has gone.

Looking at the code in main.c it seems that the device has been found and usbbootcode.bin has been successfully sent and received.  But that the re-enumeration at serial=1 is not happening.

I hacked the code a bit to step through the processes and to get debug messages which I can supply if its going to help.  The upshot is that the device detaches subsequent to the epread but doesn't re-register as expected.

I know that it may be said the cable/power may be cause.  It seems to me that getting as far through the rpiboot process as I am, seems to mitigate against a cable/power problem.  Also I have completed the Adafruit tutorial on Pi Zero gadget [https://learn.adafruit.com/turning-your-raspberry-pi-zero-into-a-usb-gadget] with the same host/cable/power and got ethernet over usb working well so I think I can say that the cable and power are both OK. 

It seems that similar problems are not being reported which leads me to think that I have made an elementary error, But I don't know what.

regards
Andy

Gadgetoid, 2016-02-07:
I've seen exactly the same problem and mentioned it to @ghollingworth, it's most probably an issue with the Pi Zero CPUID not being properly handled by the usbbootcode.bin binary. Unfortunately that binary is closed-source, so there's nothing we can do but wait patiently for a fix.

I think the lack of similar reports may simply be down to not many people even trying! But you're not alone.

AndyPi, 2016-02-08:
Thanks for that.  I guess I'll just have to wait then

Gadgetoid, 2016-03-19:
The updated `usbbootcode.bin` works on the Pi Zero, moving us one step closer. I still can't get anything to run. This may be because I have no idea what I'm doing!

kadamski, 2016-08-05:
I do know what I'm doing and I still can't get anything to run. I've tried with the kernel built and ramdisk built from buildroot, I've tried with the kernel that is actually able to run from sd-card and I've tried with bare metal example from dwelch67 which also does work from sd-card. I think it doesn't matter what I use as kernel.img since the board never leaves the SerialNumer=1 state (like it does when using msd.elf) so it has to be something wrong with usbbootcode.bin

closed 2017-06-02 by ghollingworth