raspberrypi/tools #49: Cross building with CMAKE

vanfanel, 2016-02-14:
I'm trying to cross-build a project that uses CMAKE. I am using a toolchain file for this.
First I export the path with the arm cross-compiler:

PATH=$PATH:$HOME/tools/arm-bcm2708/arm-bcm2708hardfp-linux-gnueabi/bin `

Then I have created a toolchain file with these contents (the target system usr,lib,and opt directories from Raspbian are in /opt/rpi_root):

SET(CMAKE_SYSTEM_NAME Linux)
SET(CMAKE_SYSTEM_VERSION 1)

SET(CMAKE_C_COMPILER arm-bcm2708hardfp-linux-gnueabi-gcc)
SET(CMAKE_CXX_COMPILER arm-bcm2708hardfp-linux-gnueabi-g++)

SET(CMAKE_FIND_ROOT_PATH /opt/rpi_root)

SET(CMAKE_FIND_ROOT_PATH_MODE_PROGRAM NEVER)
SET(CMAKE_FIND_ROOT_PATH_MODE_LIBRARY ONLY)
SET(CMAKE_FIND_ROOT_PATH_MODE_INCLUDE ONLY)

Then I use cmake like this, passing the toolchain file to it:

cmake -DUSE_SDL2=yes -DBUILD_TARGET=LINUX -DCMAKE_BUILD_TYPE=Release -DOGG=no -DTREMOR=NO -DOPENGL=no -DREFKEEN=no -DCMAKE_TOOLCHAIN_FILE=~/src/raspi.cmake ..

It results on the buildsystem to find everything alright, apparently:

manuel@vader:~/src/Commander-Genius/build$ cmake -DUSE_SDL2=yes -DBUILD_TARGET=LINUX -DCMAKE_BUILD_TYPE=Release -DOGG=no -DTREMOR=NO -DOPENGL=no -DREFKEEN=no -DCMAKE_TOOLCHAIN_FILE=~/src/raspi.cmake ..
-- The C compiler identification is GNU 4.7.1
-- The CXX compiler identification is GNU 4.7.1
-- Check for working C compiler: /home/manuel/tools/arm-bcm2708/arm-bcm2708hardfp-linux-gnueabi/bin/arm-bcm2708hardfp-linux-gnueabi-gcc
-- Check for working C compiler: /home/manuel/tools/arm-bcm2708/arm-bcm2708hardfp-linux-gnueabi/bin/arm-bcm2708hardfp-linux-gnueabi-gcc -- works
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - done
-- Detecting C compile features
-- Detecting C compile features - done
-- Check for working CXX compiler: /home/manuel/tools/arm-bcm2708/arm-bcm2708hardfp-linux-gnueabi/bin/arm-bcm2708hardfp-linux-gnueabi-g++
-- Check for working CXX compiler: /home/manuel/tools/arm-bcm2708/arm-bcm2708hardfp-linux-gnueabi/bin/arm-bcm2708hardfp-linux-gnueabi-g++ -- works
-- Detecting CXX compiler ABI info
-- Detecting CXX compiler ABI info - done
-- Detecting CXX compile features
-- Detecting CXX compile features - done
-- cotire 1.7.6 loaded.
-- Preparing the Build-System for Commander Genius
-- Setting SYSTEM_DATA_DIR to /usr/local/
-- Found PkgConfig: /usr/bin/pkg-config (found version "0.28")
-- checking for one of the modules 'sdl2'
-- Using shared SDL Version 2 for Commander Genius
-- Boost version: 1.55.0
-- checking for one of the modules 'SDL2_image>=2.0.0'
-- Using shared SDL Version 2 for Commander Genius
-- DOSBox-Fusion: OFF
-- Internal Downloader: OFF
-- Refkeen for Keen Dreams: no
-- CXX target CGeniusExe cotired.
-- CPACK_PACKAGE_VERSION = 1.9.1-Beta
-- Build system is prepared. To Build the project just type "make"
-- If you want to create the installation package just type "make package" after you build the project
-- Configuring done
-- Generating done
-- Build files have been written to: /home/manuel/src/Commander-Genius/build

BUT when I try to build it, there are problems with header file location:

[ 14%] Building CXX object src/graphics/CMakeFiles/graphics.dir/effects/CColorMerge.cpp.o
In file included from /usr/include/SDL2/SDL_stdinc.h:40:0,
from /usr/include/SDL2/SDL_main.h:25,
from /usr/include/SDL2/SDL.h:32,
from /home/manuel/src/Commander-Genius/src/graphics/effects/CColorMerge.h:14,
from /home/manuel/src/Commander-Genius/src/graphics/effects/CColorMerge.cpp:8:
/opt/rpi_root/usr/include/stdlib.h:760:34: fatal error: bits/stdlib-bsearch.h: No such file or directory
compilation terminated.
src/graphics/CMakeFiles/graphics.dir/build.make:54: recipe for target 'src/graphics/CMakeFiles/graphics.dir/effects/CColorMerge.cpp.o' failed
make[2]: *** [src/graphics/CMakeFiles/graphics.dir/effects/CColorMerge.cpp.o] Error 1
CMakeFiles/Makefile2:1213: recipe for target 'src/graphics/CMakeFiles/graphics.dir/all' failed
make[1]: *** [src/graphics/CMakeFiles/graphics.dir/all] Error 2
Makefile:136: recipe for target 'all' failed
make: *** [all] Error 2

That particular file is present in the target root system:

manuel@vader:~/src/Commander-Genius/build$ find /opt/rpi_root/usr/include -name stdlib-bsearch.h
/opt/rpi_root/usr/include/arm-linux-gnueabihf/bits/stdlib-bsearch.h

...but it seems the buildsystem does not look for headers in that arm-linux-gnueabihf subdirectory.
How can I convince it to do so, to look in the arm-linux-gnueabihf subdir too?

robiwano, 2018-10-29:
> That particular file is present in the target root system:
>
> manuel@vader:~/src/Commander-Genius/build$ find /opt/rpi_root/usr/include -name stdlib-bsearch.h [...]

You say "target root". Can it be that it isn't available in the host root ? Have you rsynced /usr and /lib to the host system ?

robiwano, 2018-10-29:
You might need to add /opt/rpi_root/usr/include/arm-linux-gnueabihf/ to the CMAKE_FIND_ROOT_PATH paths ?