raspberrypi/tools #51: Please provide us with a current gcc version + CMake sample for Mac, Linux, Windows

MikeMitterer, 2016-03-08:
Hi I know you are doing a great job but at the moment it's really a mess if you
want to cross-develop - specially if you are on a Mac.

What I found so far is 
http://www.welzels.de/blog/en/arm-cross-compiling-with-mac-os-x/
http://www.jaredwolff.com/blog/cross-compiling-on-mac-osx-for-raspberry-pi/
but these sites (the libs / packages) are quite outdated.

There is also a none-eabi hombrew formula (Caskroom/cask/gcc-arm-embedded) - this just overcomplicates things if you just want the write a "normal" C++ program and not a full blown kernel.

I also tried to crosstool-ng but after installing gcc 5 and all the other things I get a link error. Drives me crazy!

popcornmix, 2016-03-08:
The recommendation is to use this gcc 4.9.3 toolchain:
https://github.com/raspberrypi/tools/tree/master/arm-bcm2708/arm-rpi-4.9.3-linux-gnueabihf/bin
and this CMake file:
https://github.com/raspberrypi/userland/blob/master/makefiles/cmake/toolchains/arm-linux-gnueabihf.cmake
Note that gcc 5 is newer than Debian Jessie and you are likely to have linking issues with libc if you try using that. 4.9 matches the gcc that ships with Jessie.

MikeMitterer, 2016-03-09:
Cool + Thanks!"

closed 2016-03-09 by MikeMitterer