raspberrypi/tools #52: internal compiler error

dj95, 2016-04-01:
When I try to crossbuild the 4.1.6 emlid kernel for the raspberry pi 2, the crosscompiler crashes with the following error.

```
lib/raid6/neon4.c: In function ‘raid6_neon4_gen_syndrome_real’:
lib/raid6/neon4.c:113:1: internal compiler error: in dwarf2out_frame_debug_adjust_cfa, at dwarf2cfi.c:1078
 }
```

After a short research I found the following launchpad bug(https://bugs.launchpad.net/ubuntu/+source/gcc-4.8/+bug/1429894), which mentions the exact same problem.

popcornmix, 2016-04-01:
Which cross-compiler are you using? The newest in tools repo is:
```
arm-linux-gnueabihf-gcc --version
arm-linux-gnueabihf-gcc (crosstool-NG crosstool-ng-1.22.0-88-g8460611) 4.9.3
Copyright (C) 2015 Free Software Foundation, Inc.
This is free software; see the source for copying conditions.  There is NO
warranty; not even for MERCHANTABILITY or FITNESS FOR A PARTICULAR PURPOSE.
```

dj95, 2016-04-01:
I use the gcc-linaro x64 crosscompiler, which is delivered in version:

```
arm-linux-gnueabihf-gcc (crosstool-NG linaro-1.13.1+bzr2650 - Linaro GCC 2014.03) 4.8.3 20140303 (prerelease)
Copyright (C) 2013 Free Software Foundation, Inc.
This is free software; see the source for copying conditions.  There is NO
warranty; not even for MERCHANTABILITY or FITNESS FOR A PARTICULAR PURPOSE.
```

Downloading the latest gcc from the linaro website solved the problem. The version of this crosscompiler is
```
arm-linux-gnueabihf-gcc (Linaro GCC 5.3-2016.02) 5.3.1 20160113
Copyright (C) 2015 Free Software Foundation, Inc.
This is free software; see the source for copying conditions.  There is NO
warranty; not even for MERCHANTABILITY or FITNESS FOR A PARTICULAR PURPOSE.
```

popcornmix, 2016-04-01:
Okay, but this is the newest gcc from this repo: https://github.com/raspberrypi/tools/tree/master/arm-bcm2708/arm-rpi-4.9.3-linux-gnueabihf
I assume that doesn't have the error? (appeared to be a 4.8 error).

aelmicro, 2016-07-01:
A point of interest when trying to use the latest tools i.e. arm-rpi-4.9.3-linux-gnueabihf all compiles without and issues using existing .config from the PI made using "sudo modprobe configs", however when you change the configuration and set DEBUG_INFO=y and DEBUG_INFO_REDUCED=n then recompile this same issue raises its ugly head so may be its time to update these tools, I have yet to work out how to install the latest Linaro GCC tools as I'm a bit new to this game.

aelmicro, 2016-07-01:
Just done a "gcc --version" and it looks like something is packages incorrectly as its reporting 4.8.3 not even 4.9.3 as shown in the directory and that also in this repo so this has still got the original bug

arm-linux-gnueabihf-gcc (crosstool-NG linaro-1.13.1+bzr2650 - Linaro GCC 2014.03) 4.8.3 20140303 (prerelease)
Copyright (C) 2013 Free Software Foundation, Inc.
This is free software; see the source for copying conditions.  There is NO
warranty; not even for MERCHANTABILITY or FITNESS FOR A PARTICULAR PURPOSE.

closed 2022-01-08 by dj95