raspberrypi/tools #58: Missing license information

betafive, 2016-05-02:
There is already an issue open for adding a readme file (#10) and that would be nice to have. However, what is more important is to clarify the licenses of the various components in this repository. I would like to re-distribute the contents of 'mkimage' but need clarity on the license conditions.

Please add a top level COPYING or LICENSE file to let us know what we can do with this code.

Where do I find the source code of the toolchain in "arm-bcm2708/arm-rpi-4.9.3-linux-gnueabihf"?

Also, please clarify the license of mkimage/mkknlimg and mkimage/knlinfo. They simply say "Licensed under the terms of the GNU General Public License.". Is that the GPL v2, v3 or "v2 or later" or something else? It would be really helpful to include a full copy of the relevant version of the GPL in this repository.

Many Thanks,
Paul Barker

pelwell, 2016-05-18:
The source code to the toolchain is the source code to GCC and the associated utilities and libraries as at version 4.9.3, downloaded and built using crosstool-ng. @popcornmix can comment on any modifications made.

The mkknlimg and knlinfo utilities are now found in the scripts directory of the kernel tree, where they are covered by the licence found in the COPYING file. As the README now says, it is likely they aren't needed because current firmware assumes that the kernel is DT-capable (the standard RPi builds of the rpi-4.4.y tree require the use of Device Tree). The rest of the mkimage directory can probably also be deleted.

tpetazzoni, 2016-12-05:
There is also no license information for the rpiboot tool