raspberrypi/tools #60: libusb_control_transfer fails if length > 0x00C00000

AndyPi, 2016-05-17:
Hi,
experimenting with usbbootcode, buildroot, etc on pizero trying to get it to boot without sd card.

I find that the libusb_control_transfer function fails on a timeout if the message length is  greater than 0x00C0 0000.  
Is this the expected behaviour ?
regards
Andy

AndyPi, 2016-05-18:
Just noticed this issue #39 rpiboot fail to upload buildroot.elf + fatimage when fatimage is larger than a few MB opened on 7 Aug 2015 by grochefort.

>First, there is no problem with msd.elf turning the device in mass storage device.
I am using a compute module with its board, problem comes with the transfer of both buildroot.elf and fatimage that fail with (large) fatimage file.

>Following readme in test_code lead to generate a default fatimage file which is 20MB.

>Here are the result I get with rpiboot and such an image :

>$ sudo ./rpiboot -v -b fatimage -x ../test_code/test_code 
Waiting for BCM2835 ...
Found serial = 0: writing file ./usbbootcode.bin
libusb_bulk_transfer returned 0
Writing 16674 bytes
libusb_bulk_transfer returned 0
Successful
Waiting for BCM2835 ...
Found serial = 1: writing file ./buildroot.elf
Adding 20635648 bytes of binary to end of elf
libusb_bulk_transfer returned 0
Writing 21609448 bytes
libusb_bulk_transfer returned -1
Failed to read correct length, returned 24

>Seems that it is not the libusb_bulk_transfer which failed first but rather libusb_control_transfer (-7 / timeout ) but the returned error code is not handle (main.c:54)

>The same command line seems to succeed when the fatimage is less than 8MB.

This is same issue but not yet addressed.  
Also @ghollingworth 
>I'm in the process of creating a new filesystem boot mechanism which means that we'll not require buildroot.elf and instead use the standard start.elf
Watch this space...

Any news
If its too hard, too low priority, or just not happening, can we know ??

regards

Andy

ghollingworth, 2016-05-18:
Still working on other boot stuff at the moment to look at it, but there was a 12M limit to the size of data that can be written.

AndyPi, 2016-05-18:
Thanks for the reply, bog standard fatimage from buildroot (using their configs) is 22M. I guess I'll have to try and see what I can jettison.  I suppose the 12M is technically necessary and not arbitrary ?  It would be very helpful if it could be enlarged to cope with vanilla fatimage.
If not Ill try to be patient !

regards
Andy

AndyPi, 2016-05-20:
Shrank the fatimage to about 10M by removing networking, multimedia and sound.  The buildroot sdcard boots fine with hdmi console visible, but the fatimage gives nothing on screen.  Is there any extra configs required to get it to boot in ram ?

AndyPi, 2016-05-23:
@ghollingworth 
I understand that you are committed to other projects at the moment and usbbootcode.bin is low priority, thats OK.  I wonder though if you could advise on whether you would expect buildroot.elf to work with an up to date buildroot kernel on Zero.  
I have a working buildroot kernel, which boots off the card on Zero, but the associated fatimage does nothing at all.  Nothing on serial console either.  I am  happy to keep plugging away looking for solution, but, maybe, you have a feeling about whether it is likely to work given the evolution of the firmware and kernel ?
regards 
Andy

ghollingworth, 2016-08-04:
Should be getting back to this is the near future

pixeliz, 2016-12-14:
Pretty please, @ghollingworth ? Would be so nice to be able to boot a Zero in device mode without an SD card.

ghollingworth, 2017-01-23:
https://github.com/raspberrypi/usbboot

This build of rpiboot should no longer have this sort of limitation

closed 2017-01-23 by ghollingworth