raspberrypi/tools #61: Some GCC errors due to relatively old tools

vanfanel, 2016-05-19:
Hi,

We have found that the gcc cross-compiler is causing some problems. For example this small program:

#include <iostream>
#include <thread>

int main( int argc, char **argv )
{
    std::thread thr( []() { std::cout << "Hello thread" << std::endl; } );
    thr.join();

    return 0;
}`

if it's compiled with 

> g++ thread.cpp -o thread -std=c++11 -lpthread -mcpu=cortex-a7

Will output this:

> pi@raspberrypi:~/src $ ./thread 
> pure virtual method called
> terminate called without an active exception
> Aborted
> 

That should not happen.

Also, some projects seem to fail building on current gcc on this repo:

https://github.com/gerstrong/Commander-Genius/issues/254

That's saying 4.9.2 because it's being retried on local, but using the 4.9.4 crosscompiler in this repo gives the same results.
Any hopes for a recent 5.x cross compiler? Building one myself is relatively easy if you can give me a working .config I can use for ct-ng, trial and error is totally crazy with these things.

`

ClaymorePT, 2016-06-01:
I would hope they followed the Arch example and provide the 6.1.1 version :/

popcornmix, 2016-06-01:
You can't use a newer compiler than 4.9.3 and cross compile for debian jessie - you get issues with the version of glibc that is installed, so we have to stick with that version as the standard.

It's pretty easy to use ct-ng to create a newer compiler if you are using a distribution with newer glibc.

ClaymorePT, 2016-06-01:
will do!