raspberrypi/tools #64: Absolute path in pthread and libc linker scripts break cross-compilation

jardon-u, 2016-07-11:
I get the following error when cross compiling my program:

```
opened script file /home/ugo/dev/tools/arm-bcm2708/gcc-linaro-arm-linux-gnueabihf-raspbian-x64/arm-linux-gnueabihf/libc/usr/lib/arm-linux-gnueabihf/libpthread.so
opened script file /home/ugo/dev/tools/arm-bcm2708/gcc-linaro-arm-linux-gnueabihf-raspbian-x64/arm-linux-gnueabihf/libc/usr/lib/arm-linux-gnueabihf/libpthread.so
attempt to open /lib/arm-linux-gnueabihf/libpthread.so.0 failed
attempt to open /usr/lib/arm-linux-gnueabihf/libpthread_nonshared.a failed
/home/ugo/dev/tools/arm-bcm2708/gcc-linaro-arm-linux-gnueabihf-raspbian-x64/bin/../lib/gcc/arm-linux-gnueabihf/4.8.3/../../../../arm-linux-gnueabihf/bin/ld: cannot find /lib/arm-linux-gnueabihf/libpthread.so.0
/home/ugo/dev/tools/arm-bcm2708/gcc-linaro-arm-linux-gnueabihf-raspbian-x64/bin/../lib/gcc/arm-linux-gnueabihf/4.8.3/../../../../arm-linux-gnueabihf/bin/ld: cannot find /usr/lib/arm-linux-gnueabihf/libpthread_nonshared.a
```
Part of cmake Toolchain:
```
SET(CMAKE_FIND_ROOT_PATH $ENV{ROOTFS_DIR} $ENV{CROSSROOT_DIR})
SET(CMAKE_CXX_FLAGS  "${CMAKE_CXX_FLAGS} -march=armv6 -mtune=arm1176jzf-s -mfloat-abi=hard -mfpu=vfp --sysroot=$ENV{ROOTFS_DIR}" CACHE STRING "cache it")
SET(CMAKE_EXE_LINKER_FLAGS "${CMAKE_EXE_LINKER_FLAGS} --sysroot=$ENV{ROOTFS_DIR}" CACHE STRING "cache it")
SET(CMAKE_SHARED_LINKER_FLAGS "${CMAKE_SHARED_LINKER_FLAGS} --sysroot=$ENV{ROOTFS_DIR}" CACHE STRING "cache it")
SET(CMAKE_MODULE_LINKER_FLAGS "${CMAKE_MODULE_LINKER_FLAGS} --sysroot=$ENV{ROOTFS_DIR}" CACHE STRING "cache it")
```
ld cannot find pthread even though the library seems to correctly match the location in sysroot.
i.e. $ENV{ROOTFS_DIR}/lib/arm-linux-gnueabihf/libpthread.so.0 exists

The solution given in the following links is **to remove the relative paths from pthread (and libc) linker scripts** (tools/arm-bcm2708/gcc-linaro-arm-linux-gnueabihf-raspbian-x64/arm-linux-gnueabihf/libc/usr/lib/arm-linux-gnueabihf/libpthread.so):
http://go-robot.blogspot.fr/2014_09_01_archive.html
http://linux.bytesex.org/cross-compiler.html
http://www.mira-project.org/MIRA-doc/CrossCompilePage.html
...

It fixed my problem. Does it make sense to send a pull request removing this absolute path on the repository ?

previn-lvt, 2018-06-25:
Any updates on this? I have seen this same issue and editing the paths also fixed it. Is there a reason these absolute paths were used in the repo?

rkfg, 2018-08-20:
For those who have the same error but changing paths doesn't fix it: if you use a Raspbian chroot for libraries and stuff (like I do) you have to change these paths there as well. I built the toolchain myself with crosstool-ng, fixed the absolute path in `${TOOLCHAIN_ROOT}/arm-rpi-linux-gnueabihf/sysroot/usr/lib/libpthread.so` and it was NOT enough (or rather, it might've been pointless), but it linked after I did the same in `${RASPBIAN_CHROOT}/usr/lib/arm-linux-gnueabihf/libpthread.so` and `${RASPBIAN_CHROOT}/usr/lib/arm-linux-gnueabihf/libc.so`.

RDL-28, 2022-04-03:
It is not necessary to specify the full path, just go up a level: (../)

for libc.so
```
/* GNU ld script
   Use the shared library, but some functions are only in
   the static library, so try that secondarily.  */
OUTPUT_FORMAT(elf64-x86-64)
GROUP ( ../lib64/libc.so.6 /usr/lib64/libc_nonshared.a  AS_NEEDED ( ../lib64/ld-linux-x86-64.so.2 ) )
```
for libpthread.so
```
/* GNU ld script
   Use the shared library, but some functions are only in
   the static library, so try that secondarily.  */
OUTPUT_FORMAT(elf64-x86-64)
GROUP ( ../lib/libpthread.so.0 ../lib64/libpthread_nonshared.a )
```