raspberrypi/tools #65: Patch failling

imuguruza, 2016-08-03:
when I execute `patch -p1 < ../tools/usbboot/buildroot.patch` it fails:
```
$ patch -p1 < ../tools/usbboot/buildroot.patch
patching file board/raspberrypi/usb_test/busybox.config
patching file board/raspberrypi/usb_test/linux.config
patching file board/raspberrypi/usb_test/post_build.sh
patching file configs/raspberrypi_defconfig
Hunk #1 FAILED at 4.
Hunk #2 FAILED at 20.
2 out of 2 hunks FAILED -- saving rejects to file configs/raspberrypi_defconfig.rej
```

popcornmix, 2016-08-03:
@ghollingworth - this is something of yours isn't it?

ghollingworth, 2016-08-04:
Yes it hasn't been updated to track latest buildroot...  Common issue with distributing patches!

I'd suggest hand applying the changes in configs/raspberrypi_defconfig and then submitting a new patch

Gordon

ghollingworth, 2016-08-04:
Actually I'll leave this open, I've got to get back to this soon anyway along with the other boot modes this is part of my next batch of work

ghollingworth, 2017-01-24:
Use new rpiboot from github.com/raspberrypi/usbboot shouldn't need this anymore...

closed 2017-01-24 by ghollingworth