raspberrypi/tools #67: 4.9.3 toolchain for Mac

MikeMitterer, 2016-10-17:
Please!!!! provide us with a current toolchain for Mac. After 3 days I gave up to build one by myself. 
ct-ng with clang on Mac is horrible!!!!

MikeMitterer, 2016-12-12:
If you can't give us a Mac-Version - please provide at least the build-scripts / .config you used to build the 9.3 Linux version.

MikeMitterer, 2016-12-12:
Sorry - found the config.

MikeMitterer, 2017-01-04:
Again - 1 day spent with this mess! .config did not work on Mac.

pelwell, 2017-01-04:
How are you working around the filename case issues on the Mac? The Mac is traditionally case-preserving but case-insensitive, which causes collisions with the Linux kernel source.

Why not install a Linux image under VirtualBox? You'll probably find it to be less hassle.

carlonluca, 2017-01-05:
I typically use a Linux virtual machine as @pelwell suggests for the Pi. Another option is to use the case-sensitive filesystem for the mac, the support is built in.

MikeMitterer, 2017-01-11:
I made it myself: https://github.com/MikeMitterer/xc-gcc_4.9.3_mac_rpi

What really f...s me is that this should come from you??? the RP-foundation??? 
Every developer who wants a cross-compiler for Mac must reinvent the wheel and compile the compiler on his/her own machine. And believe me - setting up ct-ng and compiling gcc on Mac is just annoying

robindegen, 2017-04-01:
@pelwell osx can be configured to be case sensitive. It would be good enough to have that as a requirement for a first iteration i'd say.

derphilipp, 2017-04-10:
The problem with macos is that several commercial tools require a case *insensitive* file system in order to work; If it would be possible to rename the files it would be a awesome.
This also leads to many git clients on Windows and macOS (using case insensitive FS) going nuts about seemingly modified files;

robindegen, 2017-04-10:
@derphilipp Like I said, macos does support case sensitive partitioning. It's very easy to create an additional case sensitive partition specificly for this. I don't think this is too much to ask for a developer to do; if that fixes the problem.

MikeMitterer, 2017-04-29:
@robindegen Have you ever tried this yourself??? Read the forums how many developers gave up because it's so hard and time consuming. No chance without detailed and long process descriptions. AND **then main point** is it really necessary that every developer on Mac has to compile his/her own cross-compiler - it is 2017! The Raspberry foundation earns money with the Raspberry so they least what they can (and should) do is to support developers.

popcornmix, 2017-04-29:
The right solution is to use VirtualBox and install a linux OS like Ubuntu.
Linux software expects to be compiled in a Linux environment.

robindegen, 2017-04-29:
@mikemitterer yes actually. It wasn't hard at all. Only the arm compiled did not build properly with apple clang, so I build a gcc amd64 first to build the arm one. That worked. I still using this compiler.

MikeMitterer, 2017-04-29:
@robindegen You call this easy: https://www.jaredwolff.com/blog/cross-compiling-on-mac-osx-for-raspberry-pi/
Thats the best instruction I found. 

@popcornmix OMG Are you still in school? I'm asking for better/professional developer support and you give me the advice to install VirtualBox? 

I'm closing this issue because any further discussion is pointless - VirtualBox... Uahhh

closed 2017-04-29 by MikeMitterer

JamesH65, 2017-04-30:
Wow. You got professional and the best advice from a Very experienced developer and resort to childish insults.Just FYI, the majority of Pi and indeed Linux developers use either Linux or Linux in a VM. Its by far the best way. Native work on Mac or Windows in simply not worth it.

robindegen, 2017-04-30:
Yep. I'd like to second that. Though I managed to get it to work on mac just fine (it is still just unix after all)

Also, the link that you shared.. yes. I'd call that quite easy; even though the path I took was much simpler than that. If doing some basic configuration before building something scares you off, then why are you doing c++ linux development?

MikeMitterer, 2017-04-30:
Linux on Desktop is still a pain compared to Apple and don't you really think that I didn't consider switching to Linux or opening a VM or working with Docker????? My whole dev-environment is on Mac and you want me to force switching to Linux just because no one at Raspberry-Foundation is able to provide an up-to-date x-compiler for Raspberry on Mac? 

Developer support should be key for R-Foundation. The more developers, the more SW, the more devices you RF sells. 

As you can see in one of my posts from above after fiddling around for x-days I finally compiled a x-compiler for Raspi on Mac but what really pisses me off that every developer has to do this by his own. Over and over again. Someone should make a central repo to provide the compilers for all major platforms. MAC is! one of the major platforms.

MikeMitterer, 2017-04-30:
@robindegen It doesn't scare me off but it annoys me if I have to compile a compiler before I can start to work on a particular problem. We are not in the 80s anymore where you had to compile the programs on Linux yourself. 

Take Arduino for example - one of the reasons for it's success was the way you x-compile for the device. Easy, simple and on every platform.

robindegen, 2017-04-30:
@MikeMitterer Arduino is a terrible example. The AVR cross-compiler had already been around for a while when I started uni back in 2006 (so it's at least 10 years old... at the very very least), and is a very simple platform that doesn't have any kind of OS. Building a bare cross compiler for that is very easy, because you don't have to deal with kernel headers, binary interfaces and the likes. Raspberry Pi on the other hand is much newer. They're already struggling to keep the official linux version up to date (since they're still back on gcc 4).

JamesH65, 2017-04-30:
> Linux on Desktop is still a pain compared to Apple and don't you really think that I didn't consider switching to Linux or opening a VM or working with Docker????? My whole dev-environment is on Mac [...]

Linux on desktop is fine - I've been using it for years, it works fine for development, and as my every day OS. As for no-one at RPF being able to provide a compiler - well, I'm sure we can. But there is no demand for it, so it would be a waste of money|. Just You != enough demand.

> Developer support should be key for R-Foundation. The more developers, the more SW, the more devices you RF sells.

Sold over 12M so far without having specific Mac compiler support. You posts here are one of the few times it has come up in 5 years. Most people simple accept that to develop software for a Linux device, you develop on Linux. And of course, kernel development does IIRC require a Linux OS.

> As you can see in one of my posts from above after fiddling around for x-days I finally compiled a x-compiler for Raspi on Mac but what really pisses me off that every developer has to do this by [...]

Sounds like you given yourself a job there - you've done the work, now make a central repo.
Although I suspect your statement that Mac is a major platform for Linux development isn't entirely accurate. In the same way that a Windows or Linux platforms  are not a major platform for Mac development. Use the best tool for the job.

JamesH65, 2017-04-30:
> It doesn't scare me off but it annoys me if I have to compile a compiler before I can start to work on a particular problem.

What is the particular problem you are trying to solve?

robindegen, 2017-04-30:
You can also use the compiler on a linux machine over SSH from your mac. It would be virtually invisible that you're actually building on another machine. Shouldn't be too hard to set up with nfs and ssh and some clever makefiles.

MikeMitterer, 2017-04-30:
@robindegen CMake, IDE... compiling over SSH in a VM or an external PC and then sharing this again via NFS with the Raspi - really???? x-compiling and debugging over a network is hard enough but you recommend me a second machine :-)) It's getting funny now.

@JamesH65 What has the problem to do with my question for a x-compiler? Anyways - it's a lib that communicates via NR24 with other devices and communicates back to its host application via websocket. I didn't ask just for myself I asked for all the other developers out there with the same problem. Check out the forums man - people are struggling, they try other things like Docker or VMs just because it's so complicated and error prone to compile ARM-gcc with clang on Mac. 
Another story would be the age of the official supported compiler 4.9 on Raspian! People want to try out C++ 14/17 - the whole thing is way to conservative. I know there is Arch-ARM but the community is to small - but as said, thats another story

Compiling a compiler just to start working on a problem is like building a motor before you change the tires on you car. Usually on Mac I get almost every piece of SW with `brew install` - not so this x-compiler. 

And guys, really, you think it is a good, professional strategy - to convince me that I'm wrong??? 
If I develop for Android - I compile my SW and the IDE/Google toolset pushes the whole thing to the device, almost the same with iOS or with Arduino. **Raspberry is the only exception**

robindegen, 2017-04-30:
@MikeMitterer Who says the second machine can't be a VM? I've done this before. There are even various software packages that do exactly this. One is a great Visual Studio plugin called VisualGDB.

MikeMitterer, 2017-04-30:
@robindegen Overhead! Error prone - working over a second machine (VM or physical) slows the process down and adds additional problem areas. 

Thanks for really trying to help me and giving me all kind of options but I asked for a cross-compile for Raspi on Mac in an official repo. Thats all.

@JamesH65 I didn't say that Mac is a major platform for Linux development - it is one of the major platforms (Linux, Windows, Mac) 
> Linux on desktop is fine
Try Mac - it is better on Desktop. I tried all 3 OSs for quite a time and I'm really not a Mac fanboy and I like Linux but for Desktop Mac is the better choice. 

With Linux it's always the same problem (it also shows up in this discussion) every thing is free so minimal or no support is possible. "Compile it yourself - stupid ..." is still the credo (2017)!!!. And thats (one of) the reason why Linux has still a few percent market share. Server-side has taken over by the big companies but the rest is still very "garage style"

Anyways guys - thanks, but this discussion doesn't change anything so at least I have to stop here.

JamesH65, 2017-04-30:
> And guys, really, you think it is a good, professional strategy - to convince me that I'm wrong???
If I develop for Android - I compile my SW and the IDE/Google toolset pushes the whole thing to the device, almost the same with iOS or with Arduino. Raspberry is the only exception

You are wrong, we don't need to convince you, we already know the answer. Linux development is best done under Linux. Indisputable fact. Raspberry Pi is simply a Linux platform. 

Number of people developing for Linux under Mac, few. And since I am currently running the Raspberry Pi forums, I suspect that I would know if anyone was complaining about lack of Mac support. So really, why on earth should the RPF support a minority Linux dev platform?

But I think you have hit the nail on the head by mentioning Google....a multi billion $ company with huge amounts of cash. They can write tool kits for whatever platform they like with he money they find down the back of the company sofa.. And whilst it lovely to be compared with a company like Google, a 50 person outfit like Raspberry Pi (and that's about  15 engineers) cannot really compete with their billions and huge staffing. They also have a bigger sofa.