raspberrypi/tools #68: make problem Syntax error: "(" unexpected

Hudzilla55, 2016-10-19:
Using a fresh Ubuntu 15.04 installation running on VMware on OSx

Here is what I did:

sudo apt-get install bc  (was already installed)
git clone https://github.com/raspberrypi/tools
git clone --depth=1 https://github.com/raspberrypi/linux
cd linux
KERNEL=kernel7
make ARCH=arm CROSS_COMPILE=arm-linux-gnueabihf- bcm2709_defconfig
make ARCH=arm CROSS_COMPILE=arm-linux-gnueabihf-

results in the error from gcc of:
arm-bcm2708/gcc-linaro-arm-linux-gnueabihf-raspbian-x64/bin/arm-linux-gnueabihf-gcc: Syntax error: "(" unexpected

I'm seeing this has been reported in the past but no resolutions have been given.
Could someone give me a clue?

popcornmix, 2016-10-19:
What do the following report:
```
which arm-linux-gnueabihf-gcc
arm-linux-gnueabihf-gcc -v
file `which arm-linux-gnueabihf-gcc`
```

EsdrasXavier, 2019-05-03:
### Same issue here.

### Outputs:

- which arm-linux-gnueabihf-gcc
```
pi@raspberrypi:~ $ which arm-linux-gnueabihf-gcc
/usr/bin/arm-linux-gnueabihf-gcc
```
- arm-linux-gnueabihf-gcc -v
```
pi@raspberrypi:~ $ arm-linux-gnueabihf-gcc -v
Using built-in specs.
COLLECT_GCC=arm-linux-gnueabihf-gcc
COLLECT_LTO_WRAPPER=/usr/lib/gcc/arm-linux-gnueabihf/6/lto-wrapper
Target: arm-linux-gnueabihf
Configured with: ../src/configure -v --with-pkgversion='Raspbian 6.3.0-18+rpi1+deb9u1' --with-bugurl=file:///usr/share/doc/gcc-6/README.Bugs --enable-languages=c,ada,c++,java,go,d,fortran,objc,obj-c++ --prefix=/usr --program-suffix=-6 --program-prefix=arm-linux-gnueabihf- --enable-shared --enable-linker-build-id --libexecdir=/usr/lib --without-included-gettext --enable-threads=posix --libdir=/usr/lib --enable-nls --with-sysroot=/ --enable-clocale=gnu --enable-libstdcxx-debug --enable-libstdcxx-time=yes --with-default-libstdcxx-abi=new --enable-gnu-unique-object --disable-libitm --disable-libquadmath --enable-plugin --with-system-zlib --disable-browser-plugin --enable-java-awt=gtk --enable-gtk-cairo --with-java-home=/usr/lib/jvm/java-1.5.0-gcj-6-armhf/jre --enable-java-home --with-jvm-root-dir=/usr/lib/jvm/java-1.5.0-gcj-6-armhf --with-jvm-jar-dir=/usr/lib/jvm-exports/java-1.5.0-gcj-6-armhf --with-arch-directory=arm --with-ecj-jar=/usr/share/java/eclipse-ecj.jar --with-target-system-zlib --enable-objc-gc=auto --enable-multiarch --disable-sjlj-exceptions --with-arch=armv6 --with-fpu=vfp --with-float=hard --enable-checking=release --build=arm-linux-gnueabihf --host=arm-linux-gnueabihf --target=arm-linux-gnueabihf
Thread model: posix
gcc version 6.3.0 20170516 (Raspbian 6.3.0-18+rpi1+deb9u1)
```

- file `which arm-linux-gnueabihf-gcc` 
```
pi@raspberrypi:~ $ file `which arm-linux-gnueabihf-gcc`
/usr/bin/arm-linux-gnueabihf-gcc: symbolic link to gcc-6
```

If someone could help I would appreciate.

popcornmix, 2019-05-03:
It looks like you are trying to use this compile on a Pi.
arm-linux-gnueabihf-gcc is a cross compiler that runs on a x86 PC (e.g. running Ubuntu) but produces arm code.
To compile kernel on a Pi you don't need a cross compiler - just the native gcc is needed.
Follow this: https://www.raspberrypi.org/documentation/linux/kernel/building.md
(local building, not cross compiling)

EsdrasXavier, 2019-05-09:
The point is, I'm not trying to build the kernel. I was trying to compile QT5, and I was following [this guide](https://wiki.qt.io/Native_Build_of_Qt5_on_a_Raspberry_Pi). Second the guide I must use cross compile:

**In order to properly run configure for Qt5 on the Raspberry Pi, you need to point the CROSS_COMPILE variable to the native compiler located at /usr/bin/g+. The CROSS_COMPILE parameter is the path to the compiler (without the g). If you do not do this properly it will complain that it is not being cross compiled.**

But I'm not sure how to compile it and I found in somewhere in the internet were was told to user the Pi Tools to do it, I think maybe that's another issue about QT5 and not about the Pi Tools. Sorry about that