raspberrypi/tools #69: Unable to Link libbluetooth undefined reference to '__fdelt_chk@GLIBC_2.15'

JackLivio, 2016-12-01:
[Related Stack overflow question and answer here](http://stackoverflow.com/questions/40905654/cross-compile-for-arm-undefined-reference-to-fdelt-chkglibc-2-15/40914810#40914810)

I was unable to link libbluetooth due to an undefined reference to '__fdelt_chk@GLIBC_2.15'. 

I resolved this issue by using a newer release of the[ Linaro toolchain (6.1.1).](https://releases.linaro.org/components/toolchain/binaries/latest/arm-linux-gnueabihf/)

The current Linaro toolchain used by the repository is pretty old (NG-1.13.1-4.8-2014.01 ).

Any thoughts about updating the toolchain used in this repository?

danjal, 2017-03-13:
I end up with the same problem when linking to libsystemd.so.0.3.1
It has a dependency to libgcrypt.so.20 which in turn complains about undefined reference to '__fdelt_chk@GLIBC_2.15'.

popcornmix, 2017-03-13:
> Any thoughts about updating the toolchain used in this repository?

The problem with using a newer cross compiler is it causes problems with the jessie runtime libs.
E.g. If I cross compile omxplayer with the compiler you referenced, it builds okay, but running it gets:
```
omxplayer/omxplayer.bin: /usr/lib/arm-linux-gnueabihf/libstdc++.so.6: version `GLIBCXX_3.4.21' not found (required by omxplayer/omxplayer.bin)```

Sticking with gcc 4.9.3 matches the gcc version in debian jessie and avoids that issue.

danjal, 2017-03-13:
I am using the compiler under /tools/arm-bcm2708/gcc-linaro-arm-linux-gnueabihf-raspbian
I got it to work by updating some of the libraries under /tools/arm-bcm2708/gcc-linaro-arm-linux-gnueabihf-raspbian/arm-linux-gnueabihf/libc/lib/arm-linux-gnueabihf

I also tried using the compiler under /tools/arm-bcm2708/arm-rpi-4.9.3-linux-gnueabihf, but got a long list of errors.