raspberrypi/tools #71: Split usbboot in a separate repository

tpetazzoni, 2016-12-05:
Would it be possible to split the usbboot tool in a separate repository? It is a bit annoying to have to download the huge "tools" repository (with all its toolchains) only to get the usbboot tool. I recently created a package for Buildroot to build usbboot, and we don't care at all about the toolchains (since Buildroot builds its own), but having to download such a huge repository is somewhat annoying.

ghollingworth, 2016-12-05:
Actually it already has been, checkout github.com/raspberrypi/usbboot.git

But good point, I hadn't removed it from here...

tpetazzoni, 2016-12-05:
Great, thanks! So I've renamed this issue to "Remove usbboot from this repository".

ghollingworth, 2016-12-05:
Which has already been done...

tpetazzoni, 2016-12-05:
Excellent, thanks! Issue closed then!

closed 2016-12-05 by tpetazzoni