raspberrypi/tools #73: Update toolchain to support gcc 4.9 raspi3/jessie

larry104, 2017-01-23:
Can we get this toolchain updated so it is based on the current gcc version of the current Jessie release? I need to merge in libs into sysroot which are natively compiled on the current Jessie release - Unfortunately when cross-compiling I get errors like error: undefined reference to `std::__throw_out_of_range_fmt(char const*, ...)@GLIBCXX_3.4.20'  which I believe are due to the compiler gcc4.7/gcc4.9 missmatch of the toolchain.

popcornmix, 2017-01-23:
Which compiler are you using?
https://github.com/raspberrypi/tools/tree/master/arm-bcm2708/arm-rpi-4.9.3-linux-gnueabihf/bin is 4.9.

larry104, 2017-01-23:
I'm using tools/arm-bcm2708/gcc-linaro-arm-linux-gnueabihf-raspbian/bin - I was following the this tutorial https://wiki.qt.io/RaspberryPi2EGLFS#Step_by_step compile qt 5.7 - everything worked great until I needed to mix in some libs which I needed to compile on a raspi3.

popcornmix, 2017-01-23:
That is an old compiler built by linaro. They no longer make specific raspberry pi builds.
That won't be updated. I'd suggest you try the newer compiler I linked to.

larry104, 2017-01-23:
I will try to rebuild qt based on the newer compiler - thanks a lot for the quick help!

kickoune, 2017-04-14:
with the compile 4.9.2 in this github
qt not compile (./configure error)
Running configuration tests...
Checking for gold linker... 
+ cd /home/thibaud/Bureau/rpi/qt-everywhere-opensource-src-5.8.0/qtbase/config.tests && /home/thibaud/Bureau/rpi/tools/arm-bcm2708/arm-rpi-4.9.3-linux-gnueabihf/bin/arm-linux-gnueabihf-g++ -fuse-ld=gold -o conftest-out conftest.cpp
> collect2: fatal error: cannot find 'ld'
> compilation terminated.
test config.qtbase.tests.use_gold_linker FAILED
Checking for valid makespec... 
+ cd /home/thibaud/Bureau/rpi/qt-everywhere-opensource-src-5.8.0/qtbase/config.tests/common/verifyspec && /home/thibaud/Bureau/rpi/qt-everywhere-opensource-src-5.8.0/qtbase/bin/qmake -qtconf /home/thibaud/Bureau/rpi/qt-everywhere-opensource-src-5.8.0/qtbase/bin/qt.conf "CONFIG -= qt debug_and_release app_bundle lib_bundle" "CONFIG += shared console" "QMAKE_CFLAGS += --sysroot=/home/thibaud/Bureau/rpi/sysroot" "QMAKE_CXXFLAGS += --sysroot=/home/thibaud/Bureau/rpi/sysroot" "QMAKE_LFLAGS += --sysroot=/home/thibaud/Bureau/rpi/sysroot" /home/thibaud/Bureau/rpi/qt-everywhere-opensource-src-5.8.0/qtbase/config.tests/common/verifyspec
+ cd /home/thibaud/Bureau/rpi/qt-everywhere-opensource-src-5.8.0/qtbase/config.tests/common/verifyspec && MAKEFLAGS= /usr/bin/make
> /home/thibaud/Bureau/rpi/tools/arm-bcm2708/arm-rpi-4.9.3-linux-gnueabihf/bin/arm-linux-gnueabihf-g++ -c -pipe -march=armv7-a -marm -mthumb-interwork -mfpu=neon-vfpv4 -mtune=cortex-a7 -mabi=aapcs-linux -mfloat-abi=hard --sysroot=/home/thibaud/Bureau/rpi/sysroot -O2 -Wall -W -fPIC  -I. -I/home/thibaud/Bureau/rpi/qt-everywhere-opensource-src-5.8.0/qtbase/mkspecs/devices/linux-rasp-pi2-g++ -o verifyspec.o verifyspec.cpp
> /home/thibaud/Bureau/rpi/tools/arm-bcm2708/arm-rpi-4.9.3-linux-gnueabihf/bin/arm-linux-gnueabihf-g++ -Wl,-rpath-link,/home/thibaud/Bureau/rpi/sysroot/opt/vc/lib -Wl,-rpath-link,/home/thibaud/Bureau/rpi/sysroot/usr/lib/arm-linux-gnueabihf -Wl,-rpath-link,/home/thibaud/Bureau/rpi/sysroot/lib/arm-linux-gnueabihf -mfloat-abi=hard --sysroot=/home/thibaud/Bureau/rpi/sysroot -Wl,-O1 -o verifyspec verifyspec.o    
> /home/thibaud/Bureau/rpi/tools/arm-bcm2708/arm-rpi-4.9.3-linux-gnueabihf/bin/../lib/gcc/arm-linux-gnueabihf/4.9.3/../../../../arm-linux-gnueabihf/bin/ld: cannot find crt1.o: Aucun fichier ou dossier de ce type
> /home/thibaud/Bureau/rpi/tools/arm-bcm2708/arm-rpi-4.9.3-linux-gnueabihf/bin/../lib/gcc/arm-linux-gnueabihf/4.9.3/../../../../arm-linux-gnueabihf/bin/ld: cannot find crti.o: Aucun fichier ou dossier de ce type
> /home/thibaud/Bureau/rpi/tools/arm-bcm2708/arm-rpi-4.9.3-linux-gnueabihf/bin/../lib/gcc/arm-linux-gnueabihf/4.9.3/../../../../arm-linux-gnueabihf/bin/ld: cannot find -lm
> collect2: error: ld returned 1 exit status
> Makefile:64 : la recette pour la cible « verifyspec » a échouée
> make: *** [verifyspec] Erreur 1

ERROR: Cannot compile a minimal program. The toolchain or QMakeSpec is broken.

kickoune, 2017-04-14:
liner arm linux 4.9.4 official  2017 01
error in compile qt!!
L/home/thibaud/Bureau/rpi/qt-everywhere-opensource-src-5.8.0/qtbase/lib -lQt5Core -lpthread 
/home/thibaud/Bureau/rpi/qt-everywhere-opensource-src-5.8.0/qtserialbus/lib/libQt5SerialBus.so : référence indéfinie vers « QTcpServer::newConnection()@Qt_5 »
/home/thibaud/Bureau/rpi/qt-everywhere-opensource-src-5.8.0/qtserialbus/lib/libQt5SerialBus.so : référence indéfinie vers « QTcpServer::QTcpServer(QObject*)@Qt_5 »
/home/thibaud/Bureau/rpi/sysroot/usr/lib/arm-linux-gnueabihf/libQt5Network.so.5 : référence indéfinie vers « qt_safe_select(int, fd_set*, fd_set*, fd_set*, timespec const*) »
/home/thibaud/Bureau/rpi/qt-everywhere-opensource-src-5.8.0/qtserialbus/lib/libQt5SerialBus.so : référence indéfinie vers « QTcpServer::acceptError(QAbstractSocket::SocketError)@Qt_5 »
/home/thibaud/Bureau/rpi/qt-everywhere-opensource-src-5.8.0/qtserialbus/lib/libQt5SerialBus.so : référence indéfinie vers « QTcpServer::close()@Qt_5 »
/home/thibaud/Bureau/rpi/qt-everywhere-opensource-src-5.8.0/qtserialbus/lib/libQt5SerialBus.so : référence indéfinie vers « QHostAddress::~QHostAddress()@Qt_5 »
/home/thibaud/Bureau/rpi/sysroot/usr/lib/arm-linux-gnueabihf/libQt5Network.so.5 : référence indéfinie vers « qIsEffectiveTLD(QString const&) »
/home/thibaud/Bureau/rpi/sysroot/usr/lib/arm-linux-gnueabihf/libQt5Network.so.5 : référence indéfinie vers « QNonContiguousByteDevice::disableReset() »
/home/thibaud/Bureau/rpi/qt-everywhere-opensource-src-5.8.0/qtserialbus/lib/libQt5SerialBus.so : référence indéfinie vers « QAbstractSocket::connected()@Qt_5 »
/home/thibaud/Bureau/rpi/qt-everywhere-opensource-src-5.8.0/qtserialbus/lib/libQt5SerialBus.so : référence indéfinie vers « QAbstractSocket::peerName() const@Qt_5 »
/home/thibaud/Bureau/rpi/qt-everywhere-opensource-src-5.8.0/qtserialbus/lib/libQt5SerialBus.so : référence indéfinie vers « QAbstractSocket::state() const@Qt_5 »
/home/thibaud/Bureau/rpi/qt-everywhere-opensource-src-5.8.0/qtserialbus/lib/libQt5SerialBus.so : référence indéfinie vers « QAbstractSocket::staticMetaObject@Qt_5 »
/home/thibaud/Bureau/rpi/qt-everywhere-opensource-src-5.8.0/qtserialbus/lib/libQt5SerialBus.so : référence indéfinie vers « QAbstractSocket::error(QAbstractSocket::SocketError)@Qt_5 »
/home/thibaud/Bureau/rpi/qt-everywhere-opensource-src-5.8.0/qtserialbus/lib/libQt5SerialBus.so : référence indéfinie vers « QTcpServer::isListening() const@Qt_5 »
/home/thibaud/Bureau/rpi/qt-everywhere-opensource-src-5.8.0/qtserialbus/lib/libQt5SerialBus.so : référence indéfinie vers « QAbstractSocket::peerAddress() const@Qt_5 »
/home/thibaud/Bureau/rpi/qt-everywhere-opensource-src-5.8.0/qtserialbus/lib/libQt5SerialBus.so : référence indéfinie vers « QAbstractSocket::disconnected()@Qt_5 »
/home/thibaud/Bureau/rpi/qt-everywhere-opensource-src-5.8.0/qtserialbus/lib/libQt5SerialBus.so : référence indéfinie vers « QTcpServer::errorString() const@Qt_5 »
/home/thibaud/Bureau/rpi/qt-everywhere-opensource-src-5.8.0/qtserialbus/lib/libQt5SerialBus.so : référence indéfinie vers « QTcpSocket::QTcpSocket(QObject*)@Qt_5 »
/home/thibaud/Bureau/rpi/qt-everywhere-opensource-src-5.8.0/qtserialbus/lib/libQt5SerialBus.so : référence indéfinie vers « QHostAddress::QHostAddress(QString const&)@Qt_5 »
/home/thibaud/Bureau/rpi/qt-everywhere-opensource-src-5.8.0/qtserialbus/lib/libQt5SerialBus.so : référence indéfinie vers « QAbstractSocket::peerPort() const@Qt_5 »
/home/thibaud/Bureau/rpi/qt-everywhere-opensource-src-5.8.0/qtserialbus/lib/libQt5SerialBus.so : référence indéfinie vers « QTcpServer::staticMetaObject@Qt_5 »
/home/thibaud/Bureau/rpi/qt-everywhere-opensource-src-5.8.0/qtserialbus/lib/libQt5SerialBus.so : référence indéfinie vers « QTcpServer::listen(QHostAddress const&, unsigned short)@Qt_5 »
/home/thibaud/Bureau/rpi/qt-everywhere-opensource-src-5.8.0/qtserialbus/lib/libQt5SerialBus.so : référence indéfinie vers « operator<<(QDebug, QHostAddress const&)@Qt_5 »
collect2: error: ld returned 1 exit status
Makefile:90 : la recette pour la cible « ../../../bin/canbusutil » a échouée
make[4]: *** [../../../bin/canbusutil] Erreur 1
make[4] : on quitte le répertoire « /home/thibaud/Bureau/rpi/qt-everywhere-opensource-src-5.8.0/qtserialbus/src/tools/canbusutil »
Makefile:42 : la recette pour la cible « sub-canbusutil-make_first » a échouée
make[3]: *** [sub-canbusutil-make_first] Erreur 2
make[3] : on quitte le répertoire « /home/thibaud/Bureau/rpi/qt-everywhere-opensource-src-5.8.0/qtserialbus/src/tools »
Makefile:95 : la recette pour la cible « sub-tools-make_first » a échouée
make[2]: *** [sub-tools-make_first] Erreur 2
make[2] : on quitte le répertoire « /home/thibaud/Bureau/rpi/qt-everywhere-opensource-src-5.8.0/qtserialbus/src »
Makefile:44 : la recette pour la cible « sub-src-make_first » a échouée
make[1]: *** [sub-src-make_first] Erreur 2
make[1] : on quitte le répertoire « /home/thibaud/Bureau/rpi/qt-everywhere-opensource-src-5.8.0/qtserialbus »
Makefile:229 : la recette pour la cible « module-qtserialbus-make_first » a échouée
make: *** [module-qtserialbus-make_first] Erreur 2

have you a simply method for work qt 5.8 in raspberry pi.

johnmanko, 2017-05-01:
@popcornmix What device should be specified if building for RPi3?  I'm following the same guide as @larry104 (https://wiki.qt.io/RaspberryPi2EGLFS), but changed to the `arm-rpi-4.9.3-linux-gnueabihf` compiler that you recommended.

```
qtbase$ ./configure -release -opengl es2 -device linux-rasp-pi3-g++ -device-option CROSS_COMPILE=/home/user1/dev/raspberry-pi/toolchain/tools/arm-bcm2708/arm-rpi-4.9.3-linux-gnueabihf/bin/arm-linux-gnueabihf- -sysroot /home/user1/dev/raspberry-pi/toolchain/sysroot -opensource -confirm-license -make libs -prefix /usr/local/qt5pi -extprefix /home/user1/dev/raspberry-pi/toolchain/qt5pi -hostprefix /home/user1/dev/raspberry-pi/toolchain/qt5 -v
<srcbase> = /home/user1/dev/raspberry-pi/toolchain/qtbase 
<outbase> = /home/user1/dev/raspberry-pi/toolchain/qtbase 
Creating qmake...
make: Nothing to be done for 'first'.
ERROR: No device matching 'linux-rasp-pi3-g++'.

```

MarkusIppy, 2017-05-05:
for those that want to compile qt5.8 the scripts on this page work and patch the gcc on  raspi/tools for x64 Linux and build qt5.8 

https://github.com/Kukkimonsuta/rpi-buildqt

Jerwinprabu, 2017-07-04:
I am using Wiring pi for Controlling Raspberry pi GPIO. We have written a C++ programming in QT Creator. I have followed the given procedure  https://github.com/neuronalmotion/qtrpi  for QT Cross compilation. Here added some sample modules for cross compilation. I have added modules in txt file then I have added dependencies. Then tried for cross compilation. 

     I have created sample project and followed same procedure. Binary file created. When I was tried to run the exe file nothing displayed. From command line I have tried, that time wiring pi error displayed.

pi@raspberrypi:~ $ ./Test
[9;0]Unable to query physical screen size, defaulting to 100 dpi.
To override, set QT_QPA_EGLFS_PHYSICAL_WIDTH and QT_QPA_EGLFS_PHYSICAL_HEIGHT (in millimeters).
Unable to determine hardware version. I see: Hardware	: BCM2835
,
 - expecting BCM2708 or BCM2709.
If this is a genuine Raspberry Pi then please report this
to projects@drogon.net. If this is not a Raspberry Pi then you
are on your own as wiringPi is designed to support the
Raspberry Pi ONLY.
pi@raspberrypi:~ $ 

Help me to solve this issue

pelwell, 2017-07-04:
I believe wiringPi has already been updated to cope with this change. Have you got the latest source and, if so, where are you getting it from?

Jerwinprabu, 2017-07-04:
I have tried for cross compilation 

https://github.com/neuronalmotion/qtrpi/blob/master/utils/synchronize-toolchain.sh