raspberrypi/tools #75: undefined reference to `__cxa_throw_bad_array_new_length@CXXABI_1.3.8' - libicui18n.so.52

oxygen77, 2017-02-12:
Hello,

I'm trying to cross build my application on rpi2. I've followed following guide: https://wiki.qt.io/RaspberryPi2EGLFS to setup by crossbuild env.

On rpi I'm using the standard jessie updated recently.

At the end of build I get the following error:
sysroot/usr/lib/arm-linux-gnueabihf/libicui18n.so.52: undefined reference to `__cxa_throw_bad_array_new_length@CXXABI_1.3.8'

I read a previous post related to this and talking about a difference between toolchain and jessie version og libicui18n. Didn't find a way to solve this.

thx

Chris

carlonluca, 2017-02-13:
I solved by rebuilding icu with the provided gcc 4.8 and linking Qt against that build.

oxygen77, 2017-02-13:
Thx for your feedback, 

I've downloaded icu sources and plan to build it, could you direct me on how to do the cross build ? I've a linux host with build tools installed for raspberry. I've sync on the host the /lib, /usr/include and /usr/lib folder of the raspberry itself and I'm deploying qt files in /usr/local/qt5pi.

I'm targeting rpi2.

I've already build some qt examples that runs fine on the rpi, now I'm building my code which fails on link.

----- Mail original -----

carlonluca, 2017-02-13:
I wrote something about that in the past: http://thebugfreeblog.blogspot.it/2013/05/cross-building-icu-for-applications-on.html.
Or if you want you can use the build I use: http://thebugfreeblog.blogspot.it/2016/12/binaries-for-pot-550-beta1-on-qt-580.html. Download the package and extract: the build is in qtdeps.tar.

kickoune, 2017-04-14:
i have the same problem with qt5.8, i have also try to rebuild qt5.8 with the toolchain 4.9.2 in this github
i have other problem....

`Running configuration tests...
Checking for gold linker... 
+ cd /home/thibaud/Bureau/rpi/qt-everywhere-opensource-src-5.8.0/qtbase/config.tests && /home/thibaud/Bureau/rpi/tools/arm-bcm2708/arm-rpi-4.9.3-linux-gnueabihf/bin/arm-linux-gnueabihf-g++ -fuse-ld=gold -o conftest-out conftest.cpp
> collect2: fatal error: cannot find 'ld'
> compilation terminated.
test config.qtbase.tests.use_gold_linker FAILED
Checking for valid makespec... 
+ cd /home/thibaud/Bureau/rpi/qt-everywhere-opensource-src-5.8.0/qtbase/config.tests/common/verifyspec && /home/thibaud/Bureau/rpi/qt-everywhere-opensource-src-5.8.0/qtbase/bin/qmake -qtconf /home/thibaud/Bureau/rpi/qt-everywhere-opensource-src-5.8.0/qtbase/bin/qt.conf "CONFIG -= qt debug_and_release app_bundle lib_bundle" "CONFIG += shared console" "QMAKE_CFLAGS += --sysroot=/home/thibaud/Bureau/rpi/sysroot" "QMAKE_CXXFLAGS += --sysroot=/home/thibaud/Bureau/rpi/sysroot" "QMAKE_LFLAGS += --sysroot=/home/thibaud/Bureau/rpi/sysroot" /home/thibaud/Bureau/rpi/qt-everywhere-opensource-src-5.8.0/qtbase/config.tests/common/verifyspec
+ cd /home/thibaud/Bureau/rpi/qt-everywhere-opensource-src-5.8.0/qtbase/config.tests/common/verifyspec && MAKEFLAGS= /usr/bin/make
> /home/thibaud/Bureau/rpi/tools/arm-bcm2708/arm-rpi-4.9.3-linux-gnueabihf/bin/arm-linux-gnueabihf-g++ -c -pipe -march=armv7-a -marm -mthumb-interwork -mfpu=neon-vfpv4 -mtune=cortex-a7 -mabi=aapcs-linux -mfloat-abi=hard --sysroot=/home/thibaud/Bureau/rpi/sysroot -O2 -Wall -W -fPIC  -I. -I/home/thibaud/Bureau/rpi/qt-everywhere-opensource-src-5.8.0/qtbase/mkspecs/devices/linux-rasp-pi2-g++ -o verifyspec.o verifyspec.cpp
> /home/thibaud/Bureau/rpi/tools/arm-bcm2708/arm-rpi-4.9.3-linux-gnueabihf/bin/arm-linux-gnueabihf-g++ -Wl,-rpath-link,/home/thibaud/Bureau/rpi/sysroot/opt/vc/lib -Wl,-rpath-link,/home/thibaud/Bureau/rpi/sysroot/usr/lib/arm-linux-gnueabihf -Wl,-rpath-link,/home/thibaud/Bureau/rpi/sysroot/lib/arm-linux-gnueabihf -mfloat-abi=hard --sysroot=/home/thibaud/Bureau/rpi/sysroot -Wl,-O1 -o verifyspec verifyspec.o    
> /home/thibaud/Bureau/rpi/tools/arm-bcm2708/arm-rpi-4.9.3-linux-gnueabihf/bin/../lib/gcc/arm-linux-gnueabihf/4.9.3/../../../../arm-linux-gnueabihf/bin/ld: cannot find crt1.o: Aucun fichier ou dossier de ce type
> /home/thibaud/Bureau/rpi/tools/arm-bcm2708/arm-rpi-4.9.3-linux-gnueabihf/bin/../lib/gcc/arm-linux-gnueabihf/4.9.3/../../../../arm-linux-gnueabihf/bin/ld: cannot find crti.o: Aucun fichier ou dossier de ce type
> /home/thibaud/Bureau/rpi/tools/arm-bcm2708/arm-rpi-4.9.3-linux-gnueabihf/bin/../lib/gcc/arm-linux-gnueabihf/4.9.3/../../../../arm-linux-gnueabihf/bin/ld: cannot find -lm
> collect2: error: ld returned 1 exit status
> Makefile:64 : la recette pour la cible « verifyspec » a échouée
> make: *** [verifyspec] Erreur 1

ERROR: Cannot compile a minimal program. The toolchain or QMakeSpec is broken.
`

kickoune, 2017-04-14:
liner arm linux 4.9.4 official  2017 01
error in compile qt!!
L/home/thibaud/Bureau/rpi/qt-everywhere-opensource-src-5.8.0/qtbase/lib -lQt5Core -lpthread 
/home/thibaud/Bureau/rpi/qt-everywhere-opensource-src-5.8.0/qtserialbus/lib/libQt5SerialBus.so : référence indéfinie vers « QTcpServer::newConnection()@Qt_5 »
/home/thibaud/Bureau/rpi/qt-everywhere-opensource-src-5.8.0/qtserialbus/lib/libQt5SerialBus.so : référence indéfinie vers « QTcpServer::QTcpServer(QObject*)@Qt_5 »
/home/thibaud/Bureau/rpi/sysroot/usr/lib/arm-linux-gnueabihf/libQt5Network.so.5 : référence indéfinie vers « qt_safe_select(int, fd_set*, fd_set*, fd_set*, timespec const*) »
/home/thibaud/Bureau/rpi/qt-everywhere-opensource-src-5.8.0/qtserialbus/lib/libQt5SerialBus.so : référence indéfinie vers « QTcpServer::acceptError(QAbstractSocket::SocketError)@Qt_5 »
/home/thibaud/Bureau/rpi/qt-everywhere-opensource-src-5.8.0/qtserialbus/lib/libQt5SerialBus.so : référence indéfinie vers « QTcpServer::close()@Qt_5 »
/home/thibaud/Bureau/rpi/qt-everywhere-opensource-src-5.8.0/qtserialbus/lib/libQt5SerialBus.so : référence indéfinie vers « QHostAddress::~QHostAddress()@Qt_5 »
/home/thibaud/Bureau/rpi/sysroot/usr/lib/arm-linux-gnueabihf/libQt5Network.so.5 : référence indéfinie vers « qIsEffectiveTLD(QString const&) »
/home/thibaud/Bureau/rpi/sysroot/usr/lib/arm-linux-gnueabihf/libQt5Network.so.5 : référence indéfinie vers « QNonContiguousByteDevice::disableReset() »
/home/thibaud/Bureau/rpi/qt-everywhere-opensource-src-5.8.0/qtserialbus/lib/libQt5SerialBus.so : référence indéfinie vers « QAbstractSocket::connected()@Qt_5 »
/home/thibaud/Bureau/rpi/qt-everywhere-opensource-src-5.8.0/qtserialbus/lib/libQt5SerialBus.so : référence indéfinie vers « QAbstractSocket::peerName() const@Qt_5 »
/home/thibaud/Bureau/rpi/qt-everywhere-opensource-src-5.8.0/qtserialbus/lib/libQt5SerialBus.so : référence indéfinie vers « QAbstractSocket::state() const@Qt_5 »
/home/thibaud/Bureau/rpi/qt-everywhere-opensource-src-5.8.0/qtserialbus/lib/libQt5SerialBus.so : référence indéfinie vers « QAbstractSocket::staticMetaObject@Qt_5 »
/home/thibaud/Bureau/rpi/qt-everywhere-opensource-src-5.8.0/qtserialbus/lib/libQt5SerialBus.so : référence indéfinie vers « QAbstractSocket::error(QAbstractSocket::SocketError)@Qt_5 »
/home/thibaud/Bureau/rpi/qt-everywhere-opensource-src-5.8.0/qtserialbus/lib/libQt5SerialBus.so : référence indéfinie vers « QTcpServer::isListening() const@Qt_5 »
/home/thibaud/Bureau/rpi/qt-everywhere-opensource-src-5.8.0/qtserialbus/lib/libQt5SerialBus.so : référence indéfinie vers « QAbstractSocket::peerAddress() const@Qt_5 »
/home/thibaud/Bureau/rpi/qt-everywhere-opensource-src-5.8.0/qtserialbus/lib/libQt5SerialBus.so : référence indéfinie vers « QAbstractSocket::disconnected()@Qt_5 »
/home/thibaud/Bureau/rpi/qt-everywhere-opensource-src-5.8.0/qtserialbus/lib/libQt5SerialBus.so : référence indéfinie vers « QTcpServer::errorString() const@Qt_5 »
/home/thibaud/Bureau/rpi/qt-everywhere-opensource-src-5.8.0/qtserialbus/lib/libQt5SerialBus.so : référence indéfinie vers « QTcpSocket::QTcpSocket(QObject*)@Qt_5 »
/home/thibaud/Bureau/rpi/qt-everywhere-opensource-src-5.8.0/qtserialbus/lib/libQt5SerialBus.so : référence indéfinie vers « QHostAddress::QHostAddress(QString const&)@Qt_5 »
/home/thibaud/Bureau/rpi/qt-everywhere-opensource-src-5.8.0/qtserialbus/lib/libQt5SerialBus.so : référence indéfinie vers « QAbstractSocket::peerPort() const@Qt_5 »
/home/thibaud/Bureau/rpi/qt-everywhere-opensource-src-5.8.0/qtserialbus/lib/libQt5SerialBus.so : référence indéfinie vers « QTcpServer::staticMetaObject@Qt_5 »
/home/thibaud/Bureau/rpi/qt-everywhere-opensource-src-5.8.0/qtserialbus/lib/libQt5SerialBus.so : référence indéfinie vers « QTcpServer::listen(QHostAddress const&, unsigned short)@Qt_5 »
/home/thibaud/Bureau/rpi/qt-everywhere-opensource-src-5.8.0/qtserialbus/lib/libQt5SerialBus.so : référence indéfinie vers « operator<<(QDebug, QHostAddress const&)@Qt_5 »
collect2: error: ld returned 1 exit status
Makefile:90 : la recette pour la cible « ../../../bin/canbusutil » a échouée
make[4]: *** [../../../bin/canbusutil] Erreur 1
make[4] : on quitte le répertoire « /home/thibaud/Bureau/rpi/qt-everywhere-opensource-src-5.8.0/qtserialbus/src/tools/canbusutil »
Makefile:42 : la recette pour la cible « sub-canbusutil-make_first » a échouée
make[3]: *** [sub-canbusutil-make_first] Erreur 2
make[3] : on quitte le répertoire « /home/thibaud/Bureau/rpi/qt-everywhere-opensource-src-5.8.0/qtserialbus/src/tools »
Makefile:95 : la recette pour la cible « sub-tools-make_first » a échouée
make[2]: *** [sub-tools-make_first] Erreur 2
make[2] : on quitte le répertoire « /home/thibaud/Bureau/rpi/qt-everywhere-opensource-src-5.8.0/qtserialbus/src »
Makefile:44 : la recette pour la cible « sub-src-make_first » a échouée
make[1]: *** [sub-src-make_first] Erreur 2
make[1] : on quitte le répertoire « /home/thibaud/Bureau/rpi/qt-everywhere-opensource-src-5.8.0/qtserialbus »
Makefile:229 : la recette pour la cible « module-qtserialbus-make_first » a échouée
make: *** [module-qtserialbus-make_first] Erreur 2

have you a simply method for work qt 5.8 in raspberry pi.

kickoune, 2017-06-06:
anyone can be help me???
old tools chains rapbianx64 ==> ok to compiling but icu problem
with tools chain recommanded 4.9.3 problems....
ERROR: Cannot compile a minimal program. The toolchain or QMakeSpec is broken.

what is the best way for finish this quickly...?

please

AlexanderHolzer, 2017-08-08:
@kickoune I am struggling with exactly the same problems... Have you been able to fix it?

BenKluwe, 2018-04-09:
I had the same problem with compiling Qt5.3.2. It stems from an outdated stdc++ library in the gcc linaro toolchain, the version I downloaded is 4.8.3. Note that just recompiling libicu would NOT have solved this issue for me, as Qt went on to complain in a different section its code.

I solved it by overwriting the file in tools/arm-bcm2708/gcc-linaro-arm-linux-gnueabihf-raspbian/arm-linux-gnueabihf/lib/libstdc++.so.6.0.19 with the provided sysroot/usr/lib/arm-linux-gnueabihf/libstdc++.so.6.0.20. Don't forget to update the symbolic links in the toolchain too!

To figure out if the symbol is defined you can run `objdump -xrTC libstdc++.so` and search through the output.

to recap, what I did to fix the issue was:
```
pushd ${toolchain}/tools/arm-bcm2708/gcc-linaro-arm-linux-gnueabihf-raspbian/arm-linux-gnueabihf/lib/
mv libstdc++.so.6.0.19 libstdc++.so.6.0.19_backup
cp ${sysroot}/usr/lib/arm-linux-gnueabihf/libstdc++.so.6.0.20
rm libstdc++.so libstdc++.so.6
ln -s libstdc++.so.6.0.20 libstdc++.so.6
ln -s libstdc++.so.6.0.20 libstdc++.so
popd
```
Think before you run this, the libstdc++ version you have in your sysroot might not be 6.0.20!

zabdielfer, 2018-09-09:
i did that   I dont work  dammmit

sajeevank, 2019-03-26:
Hi carlonluca,

I tried to build the source code of Libicu as per the link - https://github.com/carlonluca/pi/blob/master/piomxtextures_tools/notes_icu.txt. 

But I am getting the following error (for compiling in the host):

checking whether we are cross compiling... configure: error: in `/home/sajeevank/rpi_source/libicu/build_icu_linux':
configure: error: cannot run C compiled programs.

Details given below.  

sajeevank@sajeevank-Vostro-3559:~/rpi_source/libicu/build_icu_linux$ export CPPFLAGS="-O3 -fno-short-wchar -DU_USING_ICU_NAMESPACE=1 -fno-short-enums -DU_HAVE_NL_LANGINFO_CODESET=0 -D__STDC_INT64__ -DU_TIMEZONE=0 -DUCONFIG_NO_LEGACY_CONVERSION=1 -DUCONFIG_NO_BREAK_ITERATION=1 -DUCONFIG_NO_COLLATION=1 -DUCONFIG_NO_FORMATTING=1 -DUCONFIG_NO_TRANSLITERATION=0 -DUCONFIG_NO_REGULAR_EXPRESSIONS=1"
sajeevank@sajeevank-Vostro-3559:~/rpi_source/libicu/build_icu_linux$ sh $ICU_SOURCES/source/runConfigureICU Linux --prefix=$PWD/icu_build --enable-extras=no --enable-strict=no -enable-static --enable-shared=no --enable-tests=no --enable-samples=no --enable-dyload=no
export CPP= CC=/opt/gcc-linaro-4.9.4-2017.01-x86_64_arm-linux-gnueabihf/bin/arm-linux-gnueabihf-gcc CXX=/opt/gcc-linaro-4.9.4-2017.01-x86_64_arm-linux-gnueabihf/bin/arm-linux-gnueabihf-g++ CPPFLAGS=-O3 -fno-short-wchar -DU_USING_ICU_NAMESPACE=1 -fno-short-enums -DU_HAVE_NL_LANGINFO_CODESET=0 -D__STDC_INT64__ -DU_TIMEZONE=0 -DUCONFIG_NO_LEGACY_CONVERSION=1 -DUCONFIG_NO_BREAK_ITERATION=1 -DUCONFIG_NO_COLLATION=1 -DUCONFIG_NO_FORMATTING=1 -DUCONFIG_NO_TRANSLITERATION=0 -DUCONFIG_NO_REGULAR_EXPRESSIONS=1 CFLAGS=-O3  CXXFLAGS=-O3  LDFLAGS= MAKE=
Running ./configure  --prefix=/home/sajeevank/rpi_source/libicu/build_icu_linux/icu_build --enable-extras=no --enable-strict=no -enable-static --enable-shared=no --enable-tests=no --enable-samples=no --enable-dyload=no for Linux using the clang or else GNU C++ compiler

checking for ICU version numbers... release 58.1, library 58.1, unicode version 9.0
checking build system type... x86_64-unknown-linux-gnu
checking host system type... x86_64-unknown-linux-gnu
checking whether to build debug libraries... no
checking whether to build release libraries... yes
checking whether the C compiler works... yes
checking for C compiler default output file name... a.out
checking for suffix of executables... 
checking whether we are cross compiling... configure: error: in `/home/sajeevank/rpi_source/libicu/build_icu_linux':
configure: error: cannot run C compiled programs.
If you meant to cross compile, use `--host'.
See `config.log' for more details
/home/sajeevank/rpi_source/libicu/release-58-1/source/runConfigureICU: ./configure failed
sajeevank@sajeevank-Vostro-3559:~/rpi_source/libicu/build_icu_linux$ 

I think this is some basic issue. But I am a beginner in this field. Please suggest some workaround.

Sajeevan.K

carlonluca, 2019-03-26:
I suggest you don't use the old toolchain anymore. Use a newer toolchain.

sajeevank, 2019-03-27:
Hi,

Thank You very much for the reply.

But here the toolchain is not coming into picture. Because now I am compiling for host only, not for the target RPI. right? I am failing in the first step itself.

As per my understanding there are two steps involved.

First we need to compile libicu for the host

And the second cross compile libicu for the target.

Where I am facing problem is in the first step. I think toolchain is coming only in the second step.

sajeevank, 2019-03-27:
Hi,
 
Now I noticed the lines -- 
export CPP= CC=/opt/gcc-linaro-4.9.4-2017.01-x86_64_arm-linux-gnueabihf/bin/arm-linux-gnueabihf-gcc CXX=/opt/gcc-linaro-4.9.4-2017.01-x86_64_arm-linux-gnueabihf/bin/arm-linux-gnueabihf-g++ CPPFLAGS=

And the variables CC and CXX were set for old toolchain. Now I updated it with the latest. But still the result is same.

sajeevank@sajeevank-Vostro-3559:~/rpi_source/libicu/build_icu_linux$ export CPPFLAGS="-O3 -fno-short-wchar -DU_USING_ICU_NAMESPACE=1 -fno-short-enums \                                                                                                                   
-DU_HAVE_NL_LANGINFO_CODESET=0 -D__STDC_INT64__ -DU_TIMEZONE=0 \
-DUCONFIG_NO_LEGACY_CONVERSION=1 -DUCONFIG_NO_BREAK_ITERATION=1 \
-DUCONFIG_NO_COLLATION=1 -DUCONFIG_NO_FORMATTING=1 -DUCONFIG_NO_TRANSLITERATION=0 \
-DUCONFIG_NO_REGULAR_EXPRESSIONS=1"
sajeevank@sajeevank-Vostro-3559:~/rpi_source/libicu/build_icu_linux$ sh $ICU_SOURCES/source/runConfigureICU Linux --prefix=$PWD/icu_build --enable-extras=no --enable-strict=no -enable-static --enable-shared=no --enable-tests=no --enable-samples=no --enable-dyload=no
export CPP= CC=/home/sajeevank/raspi/tools/arm-bcm2708/gcc-linaro-arm-linux-gnueabihf-raspbian-x64/bin/arm-linux-gnueabihf-gcc CXX=/home/sajeevank/raspi/tools/arm-bcm2708/gcc-linaro-arm-linux-gnueabihf-raspbian-x64/bin/arm-linux-gnueabihf-g++ CPPFLAGS=-O3 -fno-short-wchar -DU_USING_ICU_NAMESPACE=1 -fno-short-enums -DU_HAVE_NL_LANGINFO_CODESET=0 -D__STDC_INT64__ -DU_TIMEZONE=0 -DUCONFIG_NO_LEGACY_CONVERSION=1 -DUCONFIG_NO_BREAK_ITERATION=1 -DUCONFIG_NO_COLLATION=1 -DUCONFIG_NO_FORMATTING=1 -DUCONFIG_NO_TRANSLITERATION=0 -DUCONFIG_NO_REGULAR_EXPRESSIONS=1 CFLAGS=-O3  CXXFLAGS=-O3  LDFLAGS= MAKE=
Running ./configure  --prefix=/home/sajeevank/rpi_source/libicu/build_icu_linux/icu_build --enable-extras=no --enable-strict=no -enable-static --enable-shared=no --enable-tests=no --enable-samples=no --enable-dyload=no for Linux using the clang or else GNU C++ compiler

checking for ICU version numbers... release 58.1, library 58.1, unicode version 9.0
checking build system type... x86_64-unknown-linux-gnu
checking host system type... x86_64-unknown-linux-gnu
checking whether to build debug libraries... no
checking whether to build release libraries... yes
checking whether the C compiler works... yes
checking for C compiler default output file name... a.out
checking for suffix of executables... 
checking whether we are cross compiling... configure: error: in `/home/sajeevank/rpi_source/libicu/build_icu_linux':
configure: error: cannot run C compiled programs.
If you meant to cross compile, use `--host'.
See `config.log' for more details
/home/sajeevank/rpi_source/libicu/release-58-1/source/runConfigureICU: ./configure failed
sajeevank@sajeevank-Vostro-3559:~/rpi_source/libicu/build_icu_linux$ 

My doubt is when I am compiling for host, how CC and CXX are considered?

sajeevank, 2019-03-27:
Hi,
Sorry for my ignorance.
I set the variables - CC - /usr/bin/gcc and CXX - /usr/bin/g++. Now it is working fine for the host.

sajeevank, 2019-03-27:
Hi,

But for cross compile it is not successful. 

sajeevank@sajeevank-Vostro-3559:~/rpi_source/libicu/build_icu_rpi$ export CPPFLAGS="-O3 -fno-short-wchar -DU_USING_ICU_NAMESPACE=1 -fno-short-enums -pipe -march=armv8-a+crc -mtune=cortex-a53 -mfpu=crypto-neon-fp-armv8 -mfloat-abi=hard"
sajeevank@sajeevank-Vostro-3559:~/rpi_source/libicu/build_icu_rpi$ export CFLAGS=CPPFLAGS
sajeevank@sajeevank-Vostro-3559:~/rpi_source/libicu/build_icu_rpi$ export PATH=/home/sajeevank/raspi/tools/arm-bcm2708/gcc-linaro-arm-linux-gnueabihf-raspbian-x64/bin:$PATH
sajeevank@sajeevank-Vostro-3559:~/rpi_source/libicu/build_icu_rpi$ export LDFLAGS="-lc -lstdc++ -Wl,-rpath-link=/home/sajeevank/raspi/sysroot/usr/lib/"
sajeevank@sajeevank-Vostro-3559:~/rpi_source/libicu/build_icu_rpi$ export CC=/home/sajeevank/raspi/tools/arm-bcm2708/gcc-linaro-arm-linux-gnueabihf-raspbian-x64/bin/gcc
sajeevank@sajeevank-Vostro-3559:~/rpi_source/libicu/build_icu_rpi$ export CXX=/home/sajeevank/raspi/tools/arm-bcm2708/gcc-linaro-arm-linux-gnueabihf-raspbian-x64/bin/g++
sajeevank@sajeevank-Vostro-3559:~/rpi_source/libicu/build_icu_rpi$ $ICU_SOURCES/source/configure --with-cross-build=/home/sajeevank/rpi_source/libicu/build_icu_linux --enable-shared --host=arm-linux-gnueabihf --prefix=$PWD/icu_build
checking for ICU version numbers... release 58.1, library 58.1, unicode version 9.0
checking build system type... x86_64-unknown-linux-gnu
checking host system type... arm-unknown-linux-gnueabihf
checking whether to build debug libraries... no
checking whether to build release libraries... yes
checking for arm-linux-gnueabihf-clang... /home/sajeevank/raspi/tools/arm-bcm2708/gcc-linaro-arm-linux-gnueabihf-raspbian-x64/bin/gcc
checking whether the C compiler works... no
configure: error: in `/home/sajeevank/rpi_source/libicu/build_icu_rpi':
configure: error: C compiler cannot create executables
See `config.log' for more details
sajeevank@sajeevank-Vostro-3559:~/rpi_source/libicu/build_icu_rpi$

Any clues? 
The error message is ..
checking whether the C compiler works... no

sajeevank, 2019-03-27:
Hi,

I had done a few mistakes. CC and CXX were wrong. And export CFLAGS=$CPPFLAGS was the right.
By correcting these mistakes, I could cross compile, libicu successfully. 

I added this library to qt creator by using Add Library option. And then I compiled it is showing the error:

.pro is added with the following:

win32:CONFIG(release, debug|release): LIBS += -L$$PWD/../../rpi_source/libicu/build_icu_rpi/lib/release/ -licudata
else:win32:CONFIG(debug, debug|release): LIBS += -L$$PWD/../../rpi_source/libicu/build_icu_rpi/lib/debug/ -licudata
else:unix: LIBS += -L$$PWD/../../rpi_source/libicu/build_icu_rpi/lib/ -licudata

INCLUDEPATH += $$PWD/../../rpi_source/libicu/build_icu_rpi
DEPENDPATH += $$PWD/../../rpi_source/libicu/build_icu_rpi

:-1: warning: libicudata.so.57, needed by /home/sajeevank/raspi/sysroot/usr/local/qt5pi/lib/libQt5Core.so, may conflict with libicudata.so.58
/home/sajeevank/raspi/sysroot/usr/lib/arm-linux-gnueabihf/libicui18n.so.57:-1: error: undefined reference to `__cxa_throw_bad_array_new_length@CXXABI_1.3.8'

I think library is not properly added to qt creator. Or some more procedure is remaining.
Any ideas?

carlonluca, 2019-03-28:
Qt links to a different version of libicu.

sajeevank, 2019-03-28:
Hi,

Thank You for the reply. 
I am just a beginner. Would You please elaborate it a little.

I have now 

sajeevank@sajeevank-Vostro-3559:~/raspi$ find ./sysroot -name libicudata*
./sysroot/usr/local/qt5pi/lib/libicudata.so.58.1
./sysroot/usr/local/qt5pi/lib/libicudata.so.58
./sysroot/usr/local/qt5pi/lib/libicudata.so
./sysroot/usr/lib/arm-linux-gnueabihf/libicudata.a
./sysroot/usr/lib/arm-linux-gnueabihf/libicudata.so.57
./sysroot/usr/lib/arm-linux-gnueabihf/libicudata.so
./sysroot/usr/lib/arm-linux-gnueabihf/libicudata.so.57.1

Whether I should copy sysroot/usr/local/qt5pi/lib/libicudata.so.58.1 to /sysroot/usr/lib/arm-linux-gnueabihf/ or in reverse.  How the linking should be done?
If You elaborate the details, that would be very helpful.
I couldn't do it so far.

carlonluca, 2019-03-28:
"may conflict with": you are linking two conflicting versions. Qt already depends on libicu by itself.