raspberrypi/tools #78: windows toolchain

samimash, 2017-06-08:
can i install [the toolchain](https://github.com/raspberrypi/tools/tree/master/arm-bcm2708/gcc-linaro-arm-linux-gnueabihf-raspbian-x64) on windows 7
if i can what features i may loss(maybe simulator?)

pelwell, 2017-06-08:
Linux development is expected to be done using Linux. We recommend installing a real Linux image (I use Ubuntu) inside VirtualBox.

samimash, 2017-06-08:
excuse me friend but:
why is it considered a linux development ?
are not we writing bare-metal program for raspberry pi?
why  it has to be done on linux?
should not any pc do it?

pelwell, 2017-06-08:
We don't provide a Windows build of the toolchain - it is a Linux build. If you want to run the Linux-built toolchain you can do so under a virtualisation tool such as VirtualBox. Or you can try building the toolchain yourself on Windows - you may need to install Cygwin.

We use crosstool-ng to download, configure and build the toolchain - the configuration files we use are in the tools repo: https://github.com/raspberrypi/tools/tree/master/configs

JamesH65, 2017-06-08:
> raspberrypi/tools/tree/master/configs

In answer to the question about why Linux, the huge majority of Raspberry
Pi's runs Linux, only a tiny proportion are used as baremetal devices. And
even in the baremetal case, Linux generally provides a better development
environment.

123swk123, 2017-06-09:
you can run this shell script simply in your wsl and it should build the gcc for pi using the latest tool chains.

[build_cross_gcc.sh](https://github.com/123swk123/platforms/blob/master/pi/build_cross_gcc.sh)

samimash, 2017-06-09:
thank you all for replying
now it is clear 
i used vmware and install ubuntu 14.04 on it but when i run the command to install the toolchain this is what i get:
` git clone https://github.com/raspberrypi/tools
Cloning into 'tools'...
remote: Counting objects: 25343, done.
error: RPC failed; result=56, HTTP code = 200 MiB | 46.00 KiB/s    
fatal: The remote end hung up unexpectedly
fatal: early EOF
fatal: index-pack failed
`
the installation run for a bit of time then give this error

123swk123, 2017-06-09:
looks like you have network issue, try to disable firewall or anti-virus and try again.
also in your vmware try to use physical connection sharing, instead of NAT
if it still does not work then you can download the repository as a zip package and extract it locally.

samimash, 2017-06-09:
ok i will try that
but i do not see the zip package download option in the [page](https://github.com/raspberrypi/tools/tree/master/arm-bcm2708/gcc-linaro-arm-linux-gnueabihf-raspbian-x64)

pelwell, 2017-06-09:
Yes, it does sound like a network problem. Beware that the zip downloads have been found to be unreliable - people have reported corruption. We recommend using git clone.

samimash, 2017-06-12:
succesfully installed the toolchain 
know i am trying to do this as the [instructions](https://github.com/rsta2/uspi) say(building section)

> Then go to the lib/ directory of USPi and do:
> 
> make clean
> make
> 
> The ready build libuspi.a file should be in the lib/ directory.

but instead of getting the .a file i am getting this:
```
~/uspi/lib$ make clean
rm -f *.o *.a *.elf *.lst *.img *.cir *.map *~ 
~/uspi/lib$ make
arm-linux-gnueabihf-gcc -march=armv6j -mtune=arm1176jzf-s -mfloat-abi=hard  -Wall -Wno-psabi -fsigned-char -fno-builtin -nostdinc -nostdlib -std=gnu99 -undef -DRASPPI=1 -I ../include -O  -c -o uspilibrary.o uspilibrary.c
make: arm-linux-gnueabihf-gcc: Command not found
make: *** [uspilibrary.o] Error 127
```

123swk123, 2017-06-12:
Make sure the bin folder of the cross compiler tool chain is added to your path shell variable. 

~SwK

samimash, 2017-06-12:
> Make sure the bin folder of the cross compiler tool chain is added to your path shell variable
i did not quite understand what you mean
do you mean [this](https://www.raspberrypi.org/documentation/linux/kernel/building.md):

> You can then copy the  /tools/arm-bcm2708/gcc-linaro-arm-linux-gnueabihf-raspbian directory to a common location, and add  /tools/arm-bcm2708/gcc-linaro-arm-linux-gnueabihf-raspbian/bin to your $PATH in the .bashrc in your home directory
i can not find any .bashrc in my home directory

JamesH65, 2017-06-12:
This sort of question/answer session is much better hosted on the Raspberry Pi forums. You'll get a bigger audience.

samimash, 2017-06-12:
the libuspi.a file is  successfully generated 
thank you

closed 2017-06-12 by samimash