raspberrypi/tools #79: Internal compiler error when building Qt sources with debug config

kevinrombach, 2017-08-14:
I am trying to compile the Qt sources with debug config. Now im stuck with this error and cant find any solution...maybe somebody here can tell me what i can do to fix this? 

Btw: I am able compile the complete sources without the `-debug` flag in the configure call. 

make[3]: Leaving directory '/home/kevin/Qt/5.9.1/Src/qtbase/src/3rdparty/pcre2'
cd corelib/ && ( test -e Makefile || /home/kevin/Qt/5.9.1/Src/qtbase/bin/qmake -o Makefile /home/kevin/Qt/5.9.1/Src/qtbase/src/corelib/corelib.pro ) && make -f Makefile 
make[3]: Entering directory '/home/kevin/Qt/5.9.1/Src/qtbase/src/corelib'
/home/kevin/crosscompile/tools/arm-bcm2708/gcc-linaro-arm-linux-gnueabihf-raspbian-x64/bin/arm-linux-gnueabihf-g++ -c -include .pch/Qt5Core -pipe -march=armv7-a -marm -mthumb-interwork -mfpu=neon-vfpv4 -mtune=cortex-a7 -mabi=aapcs-linux -mfloat-abi=hard --sysroot=/home/kevin/sysroot -g -Og -std=c++11 -fvisibility=hidden -fvisibility-inlines-hidden -Wall -W -Wvla -D_REENTRANT -fPIC -DQT_NO_USING_NAMESPACE -DQT_NO_FOREACH -DQT_NO_NARROWING_CONVERSIONS_IN_CONNECT -DQT_BUILD_CORE_LIB -DQT_BUILDING_QT -DQT_NO_CAST_TO_ASCII -DQT_ASCII_CAST_WARNINGS -DQT_MOC_COMPAT -DQT_USE_QSTRINGBUILDER -DQT_DEPRECATED_WARNINGS -DQT_DISABLE_DEPRECATED_BEFORE=0x050000 -D_LARGEFILE64_SOURCE -D_LARGEFILE_SOURCE -DPCRE2_CODE_UNIT_WIDTH=16 -I. -Iglobal -I../3rdparty/harfbuzz/src -I../3rdparty/md5 -I../3rdparty/md4 -I../3rdparty/sha3 -I../3rdparty/double-conversion/include -I../3rdparty/double-conversion/include/double-conversion -I../3rdparty/forkfd -I../../include -I../../include/QtCore -I../../include/QtCore/5.9.1 -I../../include/QtCore/5.9.1/QtCore -I.moc -I../3rdparty/pcre2/src -I/home/kevin/crosscompile/openssl/include -I../../mkspecs/devices/linux-rasp-pi2-g++ -o .obj/qdatetimeparser.o tools/qdatetimeparser.cpp
tools/qdatetimeparser.cpp: In function ‘QString unquote(const QStringRef&)’:
tools/qdatetimeparser.cpp:342:1: internal compiler error: in push_minipool_fix, at config/arm/arm.c:14059
 }
 ^

This is how i configured the build: 

./configure -v -opengl es2  -eglfs -device linux-rasp-pi2-g++ -device-option CROSS_COMPILE=$RPI_TOOLCHAIN -sysroot $RPI_SYSROOT -opensource -confirm-license -optimized-qmake -reduce-exports -release -make libs -nomake examples -no-compile-examples -skip qtwayland -prefix /usr/local/qt5pi -skip qtwebkit -skip qt3d -debug -recheck

mkrus, 2018-06-26:
I hitting this now. Any fix?

JamesH65, 2018-06-26:
Not from us, we don't do compiler issues, that would be the gcc team. 

You could try finding a newer version of the compiler. Or maybe the problem is that the compiler is running put of memory, so you could try setting up swap (preferably on an external device rather than the SD card which is pretty slow and not very good for swap), or perhaps try zram?

popcornmix, 2018-06-26:
Looks like the bug is discussed here: https://gcc.gnu.org/bugzilla/show_bug.cgi?id=49423