raspberrypi/tools #81: GCC version not upgraded

APokorny, 2017-09-05:
What is the purpose of this repository?
Is it supposed to provide a cross compiler that matches the gcc --version of the current raspbian image?
Will there be updates of any of the toolchains in the repository?

There is a growing number of projects that will no longer compile with gcc 4.8 and 4.9 support will see a similar fate. I am aware that there is an awkward C++ ABI change with gcc 5.0, but that step has to be made at some point.

popcornmix, 2017-09-05:
I have been testing a newer gcc but unfortunately it failed testing.
I'm using crosstool-ng with [this configuration](http://paste.ubuntu.com/25471626/)
I can build kernel and /opt/vc/lib libraries fine, but kodi seg-faults on launch.

carlonluca, 2017-11-23:
If you want I built a cross toolchain gcc 6.3.0 for Stretch that seems to work for me for the moment. I built Qt, ffmpeg, omxplayer etc... I also tried kodi but I get an illegal instruction. It is built for armv7-a. I also built one identical for Mac OS. I uploaded the binaries [here](http://thebugfreeblog.blogspot.it/2017/11/raspbian-stretch-and-cross-toolchain.html).

robiwano, 2017-11-29:
@carlonluca How did you build it ? Did you use crosstool-ng ?

crapp, 2017-12-01:
@carlonluca please provide the build steps for your toolchain. I would like to build them myself

carlonluca, 2017-12-09:
@robiwano no, crosstool-ng was giving me problems with the paths. I didn't investigate further.
@crapp there are many guides on the web, I think I followed something like [this](http://preshing.com/20141119/how-to-build-a-gcc-cross-compiler).

MikeMitterer, 2018-01-17:
Developer support is incredible bad! Building a cross-compiler, specially on Mac, is not something every developer should have to do on his/her own. Even ESP32 (Espressif) has a x-compiler for every major OS. Raspberry-Org makes good money with their device - it should be possible to provide us with a cross-compiler for Mac, Windows + Linux.

JamesH65, 2018-01-17:
Wow, how rude. 

All internal development at RPT is done using either the Pi itself, or desktop Linux with the supplied cross compiler. We don't have any experience in house in Windows or Mac cross compiling. 

So I would suggest either doing dev work on RPI itself (using the RPI desktop, which is not crap BTW), or installing a VM with Linux and use the supplied cross compiler. It is old, but it works.

I'm sure that at some point we will release a newer x-compiler for Linux, but it requires a lot of work and even more testing (see comment above), and we are busy right now.

robiwano, 2018-01-17:
But the thing is, it doesn't, does it? Rasbian Stretch (f.i.) is built with the GCC6.3.0 toolchain vs. GCC4.9.3 which is the one available. As the ABI changed somewhere around GCC5 (?) linking to OS provided shared objects might lead into a world of hurt, and hard to debug errors.

That said, I for one do appreciate all the work you are doing at RPT, and I was able to find a x-toolchain to use on Ubuntu and also the one provided with VisualGDB, which works fine on Windows.

JamesH65, 2018-01-17:
Sorry, I was assuming people were wanting to cross compile kernel building, not 'app' building. I use the cross compiler for kernel builds all the time. However, for Pi app builds I use the Pi itself.  Generally fast enough for the stuff I do - and the guy who does the dev work on the desktop uses the Pi for all of that as well, so clearly its fine for some fairly heavy lifting.

APokorny, 2018-01-17:
I think the solution here is to tell people to use the debian or ubuntu repositories for building applications and kernels i.e. use the cross compilation packages from stretch if your are a lucky linux user or have the resilience to run the linux subsystem for windows10. Or use a VM solution.. 

I only came across this repo because a user reported issues with running our binaries... which were built with a new compiler. He told me that here I would find the official ones..

MikeMitterer, 2018-01-17:
@APokorny No - the solution is a cross-compiler for Windows and Mac! @JamesH65 I know that was rude - but as said: Raspberry Foundation made ~9M £ in 2015 (don't have the numbers for 2017), Expenses ~6.3M in 2015. 
This whole thing is a business and not just a fun project. 
Raspberry-Pi is 2 or 3 times more expensive than for example the Orange Pi. 
For this price I expect things like a cross-compiler and not someone telling me that it's fast enough to develop on the Pi or that I should use Linux/Docker/VM...

JamesH65, 2018-01-17:
For $35 you expect a cross compiler for Windows and Mac, two systems we don't support?

The RPF profits are pushed back straight in to education. There are limits to how much of the profits the trading arm can take of course. But TBH, the amount of profit made by the RPF is irrelevant to the decision on whether to spend a load of money building and supporting cross compilers for platforms we know nothing about. What more important is does that effort pay itself back? And the answer is, currently, no it doesn't. Lets say 1000 people want to work on a Mac, assuming profits per Pi of $3 (made up number), that's $3000 to build and support a cross compiler, ad infinitum. A completely inadequate amount of money for the effort involved. Now add the Windows version, perhaps 10000 people want to use the Windows cross compiler. We now have $30k to play with - STILL not enough. For example, we were just quoted $35k just to write a device driver for a USB wireless dongle. To do the cross compiler work, and SUPPORT it would be way more. 

I'd suggest using the Orange Pi cross compiler.

WayneKeenan, 2018-01-17:
Microsoft makes a a hell of a lot more money and yet Visual Studio for Mac doesn't cross compile C/C++ apps for Windows, does that mean 'Developer support is incredible bad!' there too ?

The solution is to use a VM.

MikeMitterer, 2018-01-17:
@WayneKeenan - MS ignored Mac and Linux because they were competitors, now they are loosing business to Mac and Linux and they have to support those systems...

@JamesH65 How much did you pay for Pixel??? A very conservative looking DE where everyone had XFCE, LXDE + Mate as an option - However...

33K for building a cross-compiler for Mac + Windows - are you kidding? If you pay 50$ per h than you have 660h - ~19 month!!!! to build a these two compilers.
ESP32 costs less than 5$ and they were able to provide their customers with an adequate compiler suite btw.

jakemagee, 2018-01-17:
It seems that this thread has gotten way side tracked...

@MikeMitterer, please find another more appropriate forum to discuss your opinions.  This thread is strictly for addressing technical issues regarding the cross toolchain.

I would suggest closing the issue since it appears that @APokorny has a workaround.  However, it seems that there is still active effort in updating the cross toolchain.

robiwano, 2018-01-17:
What? There's no workaround or solution, other than using carlonlucas binaries at the moment, unless I missed something? I mean, Raspbian Stretch was built with the 6.3.0  toolchain. Wouldn't it be possible to just provide that toolchain for running on Ubuntu f.i. ? I'm fine with running on a VM, as long as I have something to run. Right now I use the toolchain provided by SysProgs.

jakemagee, 2018-01-17:
I never stated that the issue currently has a solution.  And as for workarounds, I (possibly incorrectly) assumed you had a workaround based on your posts...

> I was able to find a x-toolchain to use on Ubuntu and also the one provided with VisualGDB, which works fine on Windows.

> other than using carlonlucas binaries at the moment

>  Right now I use the toolchain provided by SysProgs.

If, in fact, none of the above are useful to you (or anyone else) as a workaround then I'm sure we can find something else that can be until an updated toolchain is available.

johnmanko, 2018-01-20:
@MikeMitterer Have you tried cygwin on Windows?  Also, I've seen plenty of people using gcc on MacOS, so I'm sure you should be able to cross compile there as well.

MikeMitterer, 2018-01-21:
@johnmanko Thanks - but I **was** able to compile an older version of the cross-compiler for Raspberry on my Mac but it's a hassle. Mac uses clang (not gcc) and the filesystem is not case sensitive - these are the main problems. It's a very annoying, error prone and time consuming job. 

What I don't get and what really f... me is that every developer on Mac has to do this by his own and that the Raspberry foundation doesn't provide these tools - gcc 4.9 is a joke - we need a modern compiler!!! like gcc 6.3)

pelwell, 2018-01-21:
A modern compiler won't solve the case-insensitivity problem, which is why we recommend using a virtual machine running Linux - I do.

yoursunny, 2018-03-20:
I'm trying to build a large package on the Pi Zero running Raspbian Stretch, and I want to use distccd on a amd64 architecture Linux machine to help with the compilation. Having gcc 6.3.0 cross compiler would be incredibly useful.

omegacoleman, 2018-03-21:
This is definately an awful situation for many trying to modify the source of the firmware, but for 90% of the above you came to the wrong place.

If you are cross compiling apps for pi, you could just use the newest compiler you want, even gcc7 is okay. BUT WITH STATIC LINKING AND AN INDEPENDENTLY BUILT GCC. Like the one you built with --without-headers before  compiling glibc(in the toolchain build). This is the best way almost, as the firmware of embedded systems update much slower even than the cpp standard, let alone the compiler. 

You are mislead here because Debian packager knows nothing about this. The official debian arm-linux-gcc package is a stage two gcc, depending on a debian built glibc. It is limited to debian arm systems, and only works with the latest raspbian(with latest glibc).

Fedora is clearly smarter, so it shipped with the good gcc I described above.

Another thing is Debian has a clumsy multiarch dir structure, and it patches gcc to let it search different folders. If you find someting should be in /usr/lib end up in /usr/lib/arm-linux-gmueabihf, just copy it to where it should be, its okay.

crapp, 2018-03-21:
@omegacoleman  is right. It is really to build your own toolchain. If you link your application correct it will run without problems on the raspberry pi.

APokorny, 2018-03-21:
So if you came here because deployment on a armhf platform is your focus - and not a specific distribution (version) - you should look into building snaps with [snapcraft](https://snapcraft.io).

If you build your application as a snap ABI troubles and finding the right compiler are no longer a problem. You basically build your software against a specific libc and libstdc++ ABI - but with a compiler of your choice -  and you wont have further external dependencies that might not be available on the target system. On the target system the snap daemon will take care that ABI is available - either through the underlying distribution or shipped with snap.

A caveat snapcraft by default does not cross compile. You usually assemble your application within a build container. But since 2017 cross compilation was added for many of the build plugins - autotools/make go rust.. CMake needs a [workaround](https://forum.snapcraft.io/t/cross-compiling-packages-for-ubuntu-core/3517/2) ). Alternatively you can let [launchpad](https://kyrofa.com/posts/building-your-snap-on-device-there-s-a-better-way) take care of building on armhf and arm64.

ali1234, 2018-12-21:
@APokorny snap can't run on Pi Zero because Ubuntu does not support ARMv6. Neither does Launchpad, nor the arm cross compilers in Debian and Ubuntu They all require a minimum of ARMv7.

popcornmix, 2019-01-02:
@ali1234 I cloned your repo then cloned abe (master branch).
I ran the ``build.sh`` script: It failed with:
```
...
RUN:  /home/dom/projects/rpi-toolchain/_build/snapshots/dejagnu.git /home/dom/projects/rpi-toolchain/_build/snapshots/dejagnu.git~linaro-local_stable_rev_9c19d3d7473ef731b74f3a48fcce5a168bf701ec 9c19d3d7473ef731b74f3a48fcce5a168bf701ec
/home/dom/projects/rpi-toolchain/abe/lib/common.sh: line 64: /home/dom/projects/rpi-toolchain/_build/snapshots/dejagnu.git: Is a directory
WARNING: Previous command failed
ERROR (#172): checkout (Failed to create workdir for 9c19d3d7473ef731b74f3a48fcce5a168bf701ec)
ERROR (#54): checkout_all (Failed checkout out of dejagnu.)
ERROR (#118): perform_build_steps (Step CHECKOUT failed)
ERROR (#307): build_failure (Build process failed after 28 minutes)
```
Any ideas?

ali1234, 2019-01-02:
You should clone my repository with --recursive so that it clones and checks out the correct version of ABE. It is set up to use the exact same commit as was used to build the original manifest. I think the master branch of ABE is unstable.

ali1234, 2019-01-02:
Also `abe configure` should have asked you to install some things, at the very least you probably don't have git-new-workdir. I built on ubuntu 18.04 using only repo packages (git-new-workdir is packaged but has to be copied from /usr/share/doc to the path, the script tells you what to do iirc)

popcornmix, 2019-01-02:
I've tried building from a --recursive clone but no change in behaviour. Tried on two machines with the same result.

ali1234, 2019-01-02:
This is definitely an issue of missing git-new-workdir. It looks like the configure script doesn't exit with an error code if it finds a missing tool, so the script just continues on. You never see the error and then the build fails much later. In this case what is happeneing is `$NEWWORKDIR` is unset so instead of running the command it tries to run the directory, leading to the error you see. You should be able to confirm this by looking at `_build/hosts.conf`, which should have a line like `NEWWORKDIR=/home/al/.local/bin/git-new-workdir` or wherever you copied it. I expect it will be missing or empty...

ali1234, 2019-01-02:
Okay I fixed it. I forgot to sh -e, so untested errors were ignored.

popcornmix, 2019-01-03:
Thanks - build completes now.

I'm still not sure I trust the compiler though.

Basically all my attempts to bump to gcc 6 (and this also applies to this new build) fail to compile a usable kodi build.
Kodi crashes after a few seconds with what looks like heap corruption.
The 4.9 gcc doesn't have this problem.

Now it's possible there is a bug in kodi that somehow 4.9 doesn't trigger (and hasn't been observed on the many other platforms) but I went through the process of commenting out areas of code that were in the crashing callstack, and would always find another crash a bit later in unrelated code.

I've not seen an issue with building the kernel or userland libs with newer compilers, so far I've only seen the problem in kodi.

I wonder if there could be difference in libc or some brokenness in atomics that makes malloc/free not thread safe when building with newer gcc.

ali1234, 2019-01-03:
Can you please describe how you're building and running Kodi wth both compilers? Because as far as I am concerned the 4.9 compiler isn't capable of producing binaries that can run on Raspbian *at all*... not even hello world should work.

popcornmix, 2019-01-03:
[4.9 compiler](https://github.com/raspberrypi/tools/tree/master/arm-bcm2708/arm-rpi-4.9.3-linux-gnueabihf) works just fine for raspbian stretch. We use it to build the userland libs and apps (e.g. raspicam).

popcornmix, 2019-01-03:
```
dom@dom-XPS-13-9370:/tmp$ arm-linux-gnueabihf-gcc --version
arm-linux-gnueabihf-gcc (crosstool-NG crosstool-ng-1.22.0-88-g8460611) 4.9.3
Copyright (C) 2015 Free Software Foundation, Inc.
This is free software; see the source for copying conditions.  There is NO
warranty; not even for MERCHANTABILITY or FITNESS FOR A PARTICULAR PURPOSE.

dom@dom-XPS-13-9370:/tmp$ arm-linux-gnueabihf-gcc hello.c 
dom@dom-XPS-13-9370:/tmp$ file a.out
a.out: ELF 32-bit LSB executable, ARM, EABI5 version 1 (SYSV), dynamically linked, interpreter /lib/ld-linux-armhf.so.3, for GNU/Linux 2.6.32, not stripped
```
and a.out runs fine on raspbian stretch.

ali1234, 2019-01-04:
Where is your sysroot? You are building against the libraries in the default toolchain sysroot. If they don't perfectly match Raspbian you will get segfaults. This works with glibc because it has a stable ABI but the same is not true for all Kodi dependencies.

In order to actually build against Raspbian you need to specify `--sysroot /path/to/raspbian/root`.

popcornmix, 2019-01-04:
Lets see if I get kodi to build with an explicit sysroot set.

The existing compiler has a sysroot directory: https://github.com/raspberrypi/tools/tree/master/arm-bcm2708/arm-rpi-4.9.3-linux-gnueabihf/arm-linux-gnueabihf/sysroot

I believe it should be possible to configure gcc build (``--with-sysroot``) to use raspbian's sysroot (i.e. include raspbian libs in the sysroot dir of toolchain), so explicitly using ``--sysroot`` on every compile isn't required. I didn't get this working - is that option something you are familiar with?

ali1234, 2019-01-04:
The 4.9.3 compiler isn't compatible with Raspbian sysroots because it is built without `--enable-multiarch`. As such it won't find headers in the triplet prefixed directories that Debian and Raspbian use. It will refuse to compile even hello.c after it fails to find `sys/cdefs.h`.

ABE and ct-ng both claim to be able to build gcc against an existing sysroot which would mean you could build a toolchain on top of the real Raspbian glibc, however neither of them actually documents how to do this and I haven't managed to figure it out yet.

popcornmix, 2019-01-04:
But the 4.9.3 toolchain is the only toolchain that works for everything.
I use it to build raspbian kernel, raspian userland libs and apps and kodi and it just works.

Executables runs correctly on raspbian jessie and raspbian stretch.

I only have issues with newer versions of gcc.

ali1234, 2019-01-04:
Well, you still haven't said how you're building Kodi. The cross compile instructions they have tell you to use 4.9.3 and their build system compiles every dependency from scratch. So you are not building it for Raspbian if you do it that way. You're effectively building a cross-distribution bundle. The instructions also currently don't work - it fails to build libudev with "R_ARM_TLS_LE32 relocation not permitted in shared object" with both 4.9.3 compiler and my own compiler.

The kernel of course doesn't care what userspace libraries are in your sysroot because it doesn't use any of them at all - you can build it with a arm-none-gnueabihf bare metal compiler.

popcornmix, 2019-01-04:
This is how I build kodi. It's working on current kodi master (on ubuntu 18.10 with all required native dependencies installed)
```
MYADDONS="pvr.hts"
PREFIX=/home/dom/xbmc_holder/xbmc/build/install

set -e
    cd ~/xbmc_holder/xbmc/ \
    && \
    cd tools/depends \
    && \
    ./bootstrap \
    && \
    ./configure --host=arm-linux-gnueabihf \
        --prefix=${PREFIX} \
        --with-firmware=/opt/bcm-rootfs \
        --with-toolchain=/opt/arm-linux-gnueabihf \
        --with-platform=raspberry-pi2 \
    && \
    nice ionice make -j$(getconf _NPROCESSORS_ONLN) \
    && \
    cd ../.. \
    && \
    make -C tools/depends/target/cmakebuildsys \
    && \
    mkdir -p build && cd build \
    && \
    nice ionice make -j$(getconf _NPROCESSORS_ONLN) \
    && \
    make binary-addons ADDON_SRC_PREFIX=/home/dom/xbmc_holder ADDONS=$MYADDONS \
    && \
    make install
```
/opt/bcm-rootfs contains pi rootfs (I believe only used for /opt/vc/lib libs)
/opt/arm-linux-gnueabihf is a symlink to current cross compiler (4.9.3 from tools repo)

ali1234, 2019-01-04:
Yes, this is the cause of all your problems. Your build script is the same as the upstream instructions. The first time you run make it is to build all the dependencies of Kodi from source. The next steps then build Kodi against those libraries. Then you run it against the Raspbian system libraries, which are different. If this works at all, it's because of luck.

If you get segfaults and memory corruption it is because the library ABIs do not match. You should set up the library path to make Kodi use the libraries it was built against, then it will work. Alternatively build it against a real Raspbian sysroot - but this is not possible with the 4.9.3 compiler.

popcornmix, 2019-01-04:
But most of these libs aren't built by kodi depends. Kodi doesn't attempt to build libc/libm/librt/libdl/libpthread. Won't this be an issue when the compiler doesn't know what versions of these are being used at runtime?

ali1234, 2019-01-04:
Yes, it is a problem. To be really sure it works you also have to install the compiler default runtime since that it what you built against. The Linaro toolchains are split into three packages for this reason: gcc, sysroot, and runtime. The sysroot is the development headers and the runtime is just what you need when running programs.

You can normally get away with using slightly different versions of core libraries like glibc/pthreads because that stuff is really stable and they put loads of work into being backwards compatible. But the further you get from the core platform, the less likely it is to work.

jslauthor, 2019-01-31:
@ali1234 I really can't thank you enough. I've been struggling to cross-compile Qt 5.12.0 with EGLFS for longer than I care to admit 😅 but using your build script to compile the newer GCC finally got me to the finish line. I'm in awe watching my app run on a 7" LCD humming along, running shaders at 60fps without X11. 🙇

Mark-81, 2019-04-17:
@jslauthor might you share some details? I'm still trying to cross-compile Qt5.13 with EGLFS for Rpi3B+vc4 using the newer linaro-gcc toolchain suggested by @ali1234. I have a fresh Raspbian Lite environment (still in a loop device) where I downloaded dependencies (build-dep qt5-default, libssl1.0-dev libudev-dev libinput-dev libdbus-1-dev libglib2.0-dev libts-dev libx11-dev libx11-xcb-dev libxi-dev libasound2-dev fonts-dejavu gdbserver gdb-python2), sync-ed the sysroot with the out provided by the linaro toolchain.

But the configure tool of Qt5 doesn't find any EGLFS/OpenGL library:
```
Checking for OpenGL ES 2.0... 
Trying source 0 (type pkgConfig) of library opengl_es2 ...
+ PKG_CONFIG_SYSROOT_DIR=/home/mark/opt/sysroot PKG_CONFIG_LIBDIR=/home/mark/opt/sysroot/usr/lib/pkgconfig:/home/mark/opt/sysroot/usr/share/pkgconfig:/home/mark/opt/sysroot/usr/lib/arm-linux-gnueabihf/pkgconfig /usr/bin/pkg-config --exists --silence-errors glesv2
pkg-config did not find package.
  => source produced no result.
Trying source 1 (type makeSpec) of library opengl_es2 ...
header entry 'config.qtbase_gui.libraries.opengl_es2.headers.0' passed condition.
GLES2/gl2.h not found in [] and global paths.
  => source produced no result.
test config.qtbase_gui.libraries.opengl_es2 FAILED
```

I've copied the `/opt/vc/` libraries to `/home/mark/opt/sysroot/usr/`.

ali1234, 2019-04-17:
@Mark-81 Qt won't find the pkg-config files because of complicated reasons. I ended up writing a custom build tool (https://github.com/ali1234/pydo/) to manage all the ever growing list of workarounds in https://github.com/ali1234/rpi-ramdisk. In addition to the final ramdisk target it produces tgz files with the runtime libraries which you can just unpack on regular raspbian and add to the ld config. It constructs the sysroot directly from debs using multistrap and then applies all the workarounds. It downloads my abe toolchain binary releases from github. Then it builds the kernel (twice) and all the selected packages (eg Qt, gstreamer etc). You also get a qt-host directory that you can use for cross compiling outside rpi-ramdisk. qmake from that dir will set up the environment to compile against the rpi-ramdisk sysroot. It does produce Qt EGLFS executables that run on Pi Zero, but they are a bit slow compared to Pi 3.

Mark-81, 2019-04-18:
@ali1234 thanks for your explanation and your effort! I'm not interested in Pi Zero, but I just want to have a decent performance (using GPU) for RPi3B+ without X and with QML enabled, at least Qt5.12. I give it a try to your `pydo` to see if I can then drop the binaries to my already customized Raspbian Lite.

jslauthor, 2019-04-22:
@Mark-81 hopefully @ali1234's advice helped. I couldn't get 5.13 to compile. 🙁I compiler 5.12.0 successfully. Would love to know if you are successful with 5.13, however, since Lottie integration is an awesome new feature!

Mark-81, 2019-04-23:
@jslauthor unfortunately not: 

https://github.com/ali1234/rpi-ramdisk/issues/14
https://github.com/ali1234/rpi-ramdisk/issues/15

I spent too much time and for now I'm stuck with the old Qt 5.9.
It seem weird to me why it's so complex and it's not so repeatable. I mean, it should be reliable 100% once given:

- the target, i.e. RPi3B+ and Qt 5.x
- the host environment, say Debian 9 or Ubuntu 19.04
- the prerequisites: lists of packages needed
- the commands to issue (i.e. configure arguments)

At least we should have a working configuration. I understand that cannot cover all the needs for everyone, but there would be one reliable scenario to start with.

ali1234, 2019-10-13:
I have updated my toolchain to include the binutils multiarch patch (see https://github.com/raspberrypi/tools/issues/102#issuecomment-537639125). In order to do this I had to go back to using crosstool-ng.

Adding multiarch support to binutils allowed me to remove the `-Wl,---Wl,--unresolved-symbols=ignore-in-shared-libraries` hack, which turns out to invalidate some of the work I did on https://github.com/raspberrypi/firmware/issues/1013 - although there is still some overlinking, there is not as much as I thought.

It also lead me to rediscover [a bug in Debian libtool from 2005 that prevents cross-compiling some projects.](https://github.com/raspberrypi/firmware/issues/1013) I set the crosstool-ng to put a vanilla libtool in the toolchain. It will avoid the bug, but causes some overlinking. This isn't a really problem unless you are building inter-dependent debs. Either way, this is something that Debian needs to fix, but probably never will.

I think this is the best recipe yet - having proper multiarch support is something that even the original Linaro compiler never had as far as I can tell. I also made the scripts so it can be built inside Docker, although in the process I ran in to https://github.com/moby/moby/issues/13451 (but this doesn't prevent the build from working, it just prevents cleaning up the container afterwards.)

Toolchain is here: https://github.com/ali1234/rpi-toolchain

Pro, 2019-10-28:
I created a new repository which contains an updated toolchain with ARMv6 support.

Tested it on a Raspberry PI Zero with Raspbian Buster, and it works perfectly.

See:
https://stackoverflow.com/a/58559140/869402

And the repo:
https://github.com/Pro/raspi-toolchain

I'm happy for any feedback.

ali1234, 2019-10-28:
@Pro your toolchain has no multiarch support so it will not correctly link against Raspbian libraries. Configure gcc with multiarch support and apply the binutils patch from my repository to fix this.

To make it relocatable, built the dependencies static and configure everything with sysroot support.

ali1234, 2019-10-28:
I also recommend striping everything and removing unnecessary files as it will save over a gigabyte of space.

ali1234, 2019-10-28:
Attached, my notes on how Linaro ABE configures each component to achieve a relocatable system:

[configure-rel.txt](https://github.com/raspberrypi/tools/files/3779436/configure-rel.txt)

I've switched away from ABE back to crosstool-ng now because it was too hard to get ABE to apply the binutils patch (mainly because Linaro deleted all the public documentation about ABE.)