raspberrypi/tools #84: AArch32 bootstrap for RPi3 enabling caches but not the MMU

AntonioND, 2017-11-07:
In the AArch32 bootstrap, caches are enabled:

https://github.com/raspberrypi/tools/blob/b23276d2ef3d832b40ae3e4dbefac992539557d3/armstubs/armstub7.S#L60-L63

The MMU isn't enabled, though, so it is quite useless and those lines should probably be removed.

Also, according to the kernel docs, when booting the kernel, `The MMU must be off. Instruction cache may be on or off. Data cache must be off.`:

https://github.com/torvalds/linux/blob/8c27ceff3604b249a9efafbd1bd8b141b79e619d/Documentation/arm/Booting#L197-L200

pelwell, 2017-11-07:
Have you experienced a problem with the current stub or are you just striving for correctness?

AntonioND, 2017-11-07:
Just correctness, I was trying to understand what it was doing so that I could boot the kernel with multicore support from my own bootloader.

pelwell, 2017-11-07:
There's nothing wrong with that. You'll understand if this doesn't get fixed immediately, but it's on the list for a rainy day.

You could save us some time by answering a few questions:
1. Your comment suggested th at ARM caches are inoperative without the MMU - is that true?
   Never mind, I can answer this one myself - http://infocenter.arm.com/help/index.jsp?topic=/com.arm.doc.faqs/ka13835.html says the I cache works but not the D cache.
2. Since the I cache may be enabled on entry to the kernel, do you see any reason not to since it will probably save a few cycles?

You could save even more time by submitting a Pull Request with an updated armstub7.S to review.

AntonioND, 2017-11-07:
Sure, I actually wanted someone else to confirm what I suspected.

I guess that the I cache can be left enabled, yes.

Sure, I can create a PR.

closed 2021-07-13 by AntonioND