raspberrypi/tools #88: Bit definition wrong on bits/termios.h

dega03, 2018-02-09:
I believe the bit definition in [termios.h](https://github.com/raspberrypi/tools/blob/master/arm-bcm2708/gcc-linaro-arm-linux-gnueabihf-raspbian/arm-linux-gnueabihf/libc/usr/include/arm-linux-gnueabihf/bits/termios.h) should be in hex format, at the moment they aren't.

I tried a short program:
```
#include <termios.h>
#include <stdio.h>

int main()
{
	printf("CSTOPB=%X CLOCAL=%X, PARENB=%X PARODD=%X, CRTSCTS=%X\n", CSTOPB, CLOCAL, PARENB, PARODD, CRTSCTS );
	return 0;
}
```

and I am getting

> CSTOPB=40 CLOCAL=800, PARENB=100 PARODD=200, CRTSCTS=80000000

however in the definition there is:
```
#define CSTOPB      0000100
#define CLOCAL 0004000
#define PARENB 0000400
#define PARODD  0001000
# define CRTSCTS 020000000000
```

pelwell, 2018-02-09:
That's octal - a bit weird, but not that unusual. Are you saying the values are wrong? (Hint: they aren't)

dega03, 2018-02-09:
I was really struggling with this, thanks!

closed 2018-02-09 by dega03