raspberrypi/tools #93: Updating standart cpp library in toolchain

msaiducar, 2019-02-08:
Hi all,

I faced a problem while using this toolchain for my raspberry pi with paho cpp library. Detailed explanation can be found in [this issue](https://github.com/eclipse/paho.mqtt.cpp/issues/179).

Long story short, I need newer version of toolchain to compile my project. Is there a way to update this toolchain or are there any newer version of this repository?

carlonluca, 2019-02-09:
If you want to try, I built my own gcc 6.3.0 for linux and mac: http://thebugfreeblog.blogspot.com/2017/11/raspbian-stretch-and-cross-toolchain.html. I never had problems with it.

msaiducar, 2019-02-11:
I saw you post while searching but I couldn't make it work with Pi Zero. Did you test it with Pi Zero?

JamesH65, 2019-02-11:
If you are using the link from @carlonluca, then that toolchanin appears to be for ArmV7, not Arm6, which is what you need for the Zero.

carlonluca, 2019-02-12:
Yes, sorry. I only used it for armv7 and armv8.