raspberrypi/tools #99: Missing cc symlink

goldie83, 2019-07-23:
Hi, trying to cross-compile toybox and failing:
No arm-linux-gnueabihf-cc found

Is there a missing symbolic link in tools/arm-bcm2708/gcc-linaro-arm-linux-gnueabihf-raspbian-x64/bin/ ?

pelwell, 2019-07-23:
I don't think `arm-linux-gnueabihf-cc` is meant to exist - can you find a toolchain anywhere that includes it?

What configuration steps have you followed when preparing to build toybox?

goldie83, 2019-07-23:
Hi,
and what is with tools/arm-bcm2708/arm-bcm2708hardfp-linux-gnueabi/bin/ ? There is a arm-linux-gnueabihf-cc pointing to -gcc.
```
echo "--------------------------------------"
echo "UPDATING TOOLCHAIN https://github.com/raspberrypi/tools"
echo "--------------------------------------"
cd $HOME/tools
git reset --hard
git fetch

echo ""
echo "--------------------------------------"
echo "UPDATING SOURCES https://github.com/gportay/initramfs.git"
echo "--------------------------------------"
cd $HOME/initramfs
#git clean -xfd
#git submodule foreach --recursive git clean -xfd
git reset --hard
git submodule foreach --recursive git reset --hard
git fetch
git submodule update --remote --recursive

echo ""
echo "--------------------------------------"
echo "SETTING UP PATH TOOLCHAIN PATH"
echo "--------------------------------------"
PATH=$PATH:$HOME/tools/arm-bcm2708/gcc-linaro-arm-linux-gnueabihf-raspbian-x64/bin

echo ""
echo "--------------------------------------"
echo "CONFIGURE"
echo "--------------------------------------"
cp ../initramfs.config .config
cp ../busybox.config busybox_defconfig
cp ../toybox.config toybox_defconfig

echo ""
echo "--------------------------------------"
echo "BUILD"
echo "--------------------------------------"
make ARCH=arm CROSS_COMPILE=arm-linux-gnueabihf- LDFLAGS=-lrt

```

[configs.zip](https://github.com/raspberrypi/tools/files/3422744/configs.zip)