raspberrypi/tools #102: Upgrade raspberrypi's cross compile tool to gcc 8.3.0?

jiapei100, 2019-09-06:
Any plan to upgrade raspberrypi's cross compile tool to gcc 8.3.0? Some times, it's seriously slow to compile the source code on raspberry pi directly...

carlonluca, 2019-09-09:
If you want I uploaded mine [here](https://thebugfreeblog.blogspot.com/2019/09/raspbian-buster-gcc.html). It should be 100% compatible with Buster. I have not experienced problems with it yet.

ali1234, 2019-10-02:
@carlonluca unfortunately your toolchain is not compatible with buster: your binutils is not patched for multiarch. Your gcc is, so you get linker errors like this:

```/opt/rpi/rpi-gcc-8.3.0/lib/gcc/arm-linux-gnueabihf/8.3.0/../../../../arm-linux-gnueabihf/bin/ld: warning: libpcre.so.3, needed by /home/al/Source/rpi-ramdisk/sysroot/sysroot/usr/lib/arm-linux-gnueabihf/libglib-2.0.so, not found (try using -rpath or -rpath-link)```

In order to fix this you need to apply http://bazaar.launchpad.net/~doko/binutils/pkg-2.24-debian/view/head:/patches/129_multiarch_libpath.patch to binutils source tree before building it.

mtekeli, 2019-10-02:
Can this be of any use for this?

https://sourceforge.net/p/raspberry-pi-cross-compilers/code/ci/master/tree/

ali1234, 2019-10-02:
@mtekeli all the download links for that are broken, but I expect it has the same problem.

carlonluca, 2019-10-02:
@ali1234 thanks for the info. I didn't encounter that error with any of my builds. How did you trigger that error?

ali1234, 2019-10-02:
I tried to cross compile libgssdp from git master against a raspbian sysroot.

Pro, 2019-10-28:
I created a new repository which contains an updated toolchain with ARMv6 support.

Tested it on a Raspberry PI Zero with Raspbian Buster, and it works perfectly.

See:
https://stackoverflow.com/a/58559140/869402

And the repo:
https://github.com/Pro/raspi-toolchain

I'm happy for any feedback.

JamesH65, 2019-10-28:
Does it cross compile the kernel correctly and the result run OK? That's usually a good test for a new toolchain.

Pro, 2019-10-28:
> Does it cross compile the kernel correctly and the result run OK? That's usually a good test for a new toolchain.

I did not test that yet. Honestly, I did not yet try to compile a kernel myself. @JamesH65 do you know any good tutorial or example which shows how to do that?
I could try to include this test into Continuous Integration, then we will see :+1:

pelwell, 2019-10-28:
The official RPi kernel building guide is here: https://www.raspberrypi.org/documentation/linux/kernel/building.md

martburg, 2019-11-06:
the toolchain according to ali1234 succeded to cross-compile qt 5.11.3

carlonluca, 2021-10-08:
@mahaju this toolchain here https://thebugfreeblog.blogspot.com/2019/09/raspbian-buster-gcc.html works for every Qt version I tested, up to 5.15.1. I tested for both Linux and Mac OS.