raspberrypi/tools #108: Question: use standard Debian crosscompiler gcc-arm-linux-gnueabihf ?

rpier1, 2020-02-20:
Hi,

in this repo there is a very particular crosscompiler
https://github.com/raspberrypi/tools/tree/master/arm-bcm2708/gcc-linaro-arm-linux-gnueabihf-raspbian-x64

which is also referenced in the official Kernel building guide
https://www.raspberrypi.org/documentation/linux/kernel/building.md (under Section "Install toolchain")

**Question:**
Can I also use the standard _gcc-arm-linux-gnueabihf_ APT Package  
  https://packages.debian.org/buster/gcc-arm-linux-gnueabihf  
from Debian buster? (This is gcc version 8.3.0)

Or can there be incompatibilities and things break?

( I only want to rebuild a modified kernel, but for all other packages, I want to use Raspberry Pi's APT servers: http://raspbian.raspberrypi.org/raspbian/ and http://archive.raspberrypi.org/debian/ )

In particular: When building the kernel deb-pkg, I also get the following:   
`linux-libc-dev_..._armhf.deb`

Is it important (or essential!) that I must install this `linux-libc-dev_..._armhf.deb`, when compiling the kernel with Debian's own _gcc-arm-linux-gnueabihf_ crosscompiler ?

(What about `linux-headers-..._armhf.deb` ... , which is also a output of the kernel deb-pkg build?)

Thanks.

popcornmix, 2020-02-20:
There's nothing special about the compilers hosted here other than passing in the target flags for a Pi1 by default.
Standard debian gcc-arm-linux-gnueabihf will compile the kernel just fine.

rpier1, 2020-02-20:
OK thanks.

Is this always the case? What if I user a super new shiney gcc?

Why does it work? Does it work because the interface to the libc is standardized, and in the end it does not matter what compiler I use; as long as the machine code generated is appropriate (armhf); and the libc interface is compatible (unchanged)?

In the same vein: Above I talk about   
[_gcc-arm-linux-gnueabihf_](https://packages.debian.org/buster/gcc-arm-linux-gnueabihf) . What about  
[_gcc-arm-none-eabi_](https://packages.debian.org/buster/gcc-arm-none-eabi) ?

rpier1, 2020-02-20:
> [_gcc-arm-linux-gnueabihf_](https://packages.debian.org/buster/gcc-arm-linux-gnueabihf) . What about
> [_gcc-arm-none-eabi_](https://packages.debian.org/buster/gcc-arm-none-eabi) ?

Ah. The difference is shown here:
https://stackoverflow.com/a/38989869

"The bare-metal ABI will assume a different C library (newlib for example, or even no C library) to the Linux ABI (which assumes glibc). Therefore, the compiler may make different function calls depending on what it believes is available above and beyond the Standard C library."

popcornmix, 2020-02-20:
In general you can use any compiler. Apart from occasional compiler bugs, a latest and greatest gcc will build the kernel just fine.

libc (and libm etc) is a different matter. Cross compiling using a different version of libc may not work. Generally when there is an incompatible change you will fail to run the executable with a missing symbol error.

But that doesn't affect the kernel which doesn't use run time libraries. That should be fine.

rpier1, 2020-02-20:
Thank you for your answer.

If I interpret it correctly it means that for non-kernel programs, i.e. userland programs; I would be required to use the exact cross-compiler from this repo, since that would avoid libc (and libm) version incompatibilities.

rpier1, 2020-02-20:
PS: Do you happen to know what `linux-libc-dev_..._armhf.deb` is (result of a kernel build of target deb-pkg)?

How does that fit into the picture?

popcornmix, 2020-02-20:
> If I interpret it correctly it means that for non-kernel programs, i.e. userland programs; I would be required to use the exact cross-compiler from this repo, since that would avoid libc (and libm) version incompatibilities.

There's nothing special about the compilers here. They were just a gcc build from a while ago with different (pi1) default options for the cpu target.
They haven't been updated since raspbian updated to buster. There are probably better options to use.
They do work for building the libs and apps in [userland](https://github.com/raspberrypi/userland) even though libc/libm have moved on. You may have problems with more complicated applications.

To be honest cross-compiling large packages almost always involves substantial manual tweaking. Usually the right option is to build natively on the device unless you know you will be building the same package many times.

rpier1, 2020-02-20:
Thanks for taking the time to reply.
I'll be building the kernel with normal Debian `gcc-arm-linux-gnueabihf`.   
And indeed, I indend to build other packages "natively" nicely inside the image (qemu) using `apt-get source`.

PS: I had some specific questions about Raspberry Kernel deb packaging ([ref](https://github.com/RPi-Distro/repo/issues/164)), but will ask that one the raspberry pi forum, once I'm activated (it's taking a long time)

closed 2020-02-20 by rpier1