raspberrypi/tools #112: valgrind not working on Raspbian Buster

HankB, 2020-05-29:
hbarta@prelude:~ $ cat /etc/issue
Raspbian GNU/Linux 10 \n \l

hbarta@prelude:~ $ dpkg -l valgrind
Desired=Unknown/Install/Remove/Purge/Hold
| Status=Not/Inst/Conf-files/Unpacked/halF-conf/Half-inst/trig-aWait/Trig-pend
|/ Err?=(none)/Reinst-required (Status,Err: uppercase=bad)
||/ Name           Version        Architecture Description
+++-==============-==============-============-=============================================================
ii  valgrind       1:3.7.0-6+rpi6 armhf        instrumentation framework for building dynamic analysis tools
hbarta@prelude:~ $ 

In the simplest of programs (such as hello world in C or C++ `valgrind` installed from the Raspbian repo reports lots of errors. If the current version is downloaded from https://www.valgrind.org/downloads/current.html#current and built from current sources (3.16.0), It builds cleanly and works without reporting extraneous errors.

Please see post https://www.raspberrypi.org/forums/viewtopic.php?f=33&t=275485&e=1&view=unread#unread for further information.

carlonluca, 2020-05-29:
I confirm I had the same problem. I had to use 3.15.0 to get useful results.

JamesH65, 2020-05-29:
This is the wrong repo for this report. It's a Raspbian issue so need to go here

https://bugs.launchpad.net/raspbian/

closed 2020-05-29 by popcornmix

HankB, 2020-05-29:
Many thanks for pointing me to the correct repo to report this. For anyone following, I added my comments to https://bugs.launchpad.net/raspbian/+bug/1743293?comments=all which I believe to be the same issue.