raspberrypi/tools #118: Remove case-sensitive duplicate files

jdoubleu, 2020-10-10:
Unfortunately, git gets confused on non case-sensitive file systems (e.g. Windows NTFS by default), showing file changes, because it cannot properly distinguish between the two files.

This is only really a problem when there are two files with the same name but different cases in a folder. Please read this [StackOverflow answer](https://stackoverflow.com/a/2529872/7201175).

When I checked out this repository, I get the following changes:

<details> 
  <summary>Click to expand</summary>

```
HEAD detached at 0c39cb5
Changes not staged for commit:
  (use "git add <file>..." to update what will be committed)
  (use "git checkout -- <file>..." to discard changes in working directory)

        modified:   arm-bcm2708/arm-bcm2708-linux-gnueabi/arm-bcm2708-linux-gnueabi/sysroot/usr/include/linux/netfilter/xt_CONNMARK.h
        modified:   arm-bcm2708/arm-bcm2708-linux-gnueabi/arm-bcm2708-linux-gnueabi/sysroot/usr/include/linux/netfilter/xt_DSCP.h
        modified:   arm-bcm2708/arm-bcm2708-linux-gnueabi/arm-bcm2708-linux-gnueabi/sysroot/usr/include/linux/netfilter/xt_MARK.h
        modified:   arm-bcm2708/arm-bcm2708-linux-gnueabi/arm-bcm2708-linux-gnueabi/sysroot/usr/include/linux/netfilter/xt_RATEEST.h
        modified:   arm-bcm2708/arm-bcm2708-linux-gnueabi/arm-bcm2708-linux-gnueabi/sysroot/usr/include/linux/netfilter/xt_TCPMSS.h
        modified:   arm-bcm2708/arm-bcm2708-linux-gnueabi/arm-bcm2708-linux-gnueabi/sysroot/usr/include/linux/netfilter_ipv4/ipt_ECN.h
        modified:   arm-bcm2708/arm-bcm2708-linux-gnueabi/arm-bcm2708-linux-gnueabi/sysroot/usr/include/linux/netfilter_ipv4/ipt_TTL.h
        modified:   arm-bcm2708/arm-bcm2708-linux-gnueabi/arm-bcm2708-linux-gnueabi/sysroot/usr/include/linux/netfilter_ipv6/ip6t_HL.h
        modified:   arm-bcm2708/arm-bcm2708hardfp-linux-gnueabi/arm-bcm2708hardfp-linux-gnueabi/sysroot/usr/include/linux/netfilter/xt_CONNMARK.h
        modified:   arm-bcm2708/arm-bcm2708hardfp-linux-gnueabi/arm-bcm2708hardfp-linux-gnueabi/sysroot/usr/include/linux/netfilter/xt_DSCP.h
        modified:   arm-bcm2708/arm-bcm2708hardfp-linux-gnueabi/arm-bcm2708hardfp-linux-gnueabi/sysroot/usr/include/linux/netfilter/xt_MARK.h
        modified:   arm-bcm2708/arm-bcm2708hardfp-linux-gnueabi/arm-bcm2708hardfp-linux-gnueabi/sysroot/usr/include/linux/netfilter/xt_RATEEST.h
        modified:   arm-bcm2708/arm-bcm2708hardfp-linux-gnueabi/arm-bcm2708hardfp-linux-gnueabi/sysroot/usr/include/linux/netfilter/xt_TCPMSS.h
        modified:   arm-bcm2708/arm-bcm2708hardfp-linux-gnueabi/arm-bcm2708hardfp-linux-gnueabi/sysroot/usr/include/linux/netfilter_ipv4/ipt_ECN.h
        modified:   arm-bcm2708/arm-bcm2708hardfp-linux-gnueabi/arm-bcm2708hardfp-linux-gnueabi/sysroot/usr/include/linux/netfilter_ipv4/ipt_TTL.h
        modified:   arm-bcm2708/arm-bcm2708hardfp-linux-gnueabi/arm-bcm2708hardfp-linux-gnueabi/sysroot/usr/include/linux/netfilter_ipv6/ip6t_HL.h
        modified:   arm-bcm2708/arm-rpi-4.9.3-linux-gnueabihf/arm-linux-gnueabihf/sysroot/usr/include/linux/netfilter/xt_CONNMARK.h
        modified:   arm-bcm2708/arm-rpi-4.9.3-linux-gnueabihf/arm-linux-gnueabihf/sysroot/usr/include/linux/netfilter/xt_DSCP.h
        modified:   arm-bcm2708/arm-rpi-4.9.3-linux-gnueabihf/arm-linux-gnueabihf/sysroot/usr/include/linux/netfilter/xt_MARK.h
        modified:   arm-bcm2708/arm-rpi-4.9.3-linux-gnueabihf/arm-linux-gnueabihf/sysroot/usr/include/linux/netfilter/xt_RATEEST.h
        modified:   arm-bcm2708/arm-rpi-4.9.3-linux-gnueabihf/arm-linux-gnueabihf/sysroot/usr/include/linux/netfilter/xt_TCPMSS.h
        modified:   arm-bcm2708/arm-rpi-4.9.3-linux-gnueabihf/arm-linux-gnueabihf/sysroot/usr/include/linux/netfilter_ipv4/ipt_ECN.h
        modified:   arm-bcm2708/arm-rpi-4.9.3-linux-gnueabihf/arm-linux-gnueabihf/sysroot/usr/include/linux/netfilter_ipv4/ipt_TTL.h
        modified:   arm-bcm2708/arm-rpi-4.9.3-linux-gnueabihf/arm-linux-gnueabihf/sysroot/usr/include/linux/netfilter_ipv6/ip6t_HL.h
        modified:   arm-bcm2708/gcc-linaro-arm-linux-gnueabihf-raspbian-x64/arm-linux-gnueabihf/libc/usr/include/linux/netfilter/xt_CONNMARK.h
        modified:   arm-bcm2708/gcc-linaro-arm-linux-gnueabihf-raspbian-x64/arm-linux-gnueabihf/libc/usr/include/linux/netfilter/xt_DSCP.h
        modified:   arm-bcm2708/gcc-linaro-arm-linux-gnueabihf-raspbian-x64/arm-linux-gnueabihf/libc/usr/include/linux/netfilter/xt_MARK.h
        modified:   arm-bcm2708/gcc-linaro-arm-linux-gnueabihf-raspbian-x64/arm-linux-gnueabihf/libc/usr/include/linux/netfilter/xt_RATEEST.h
        modified:   arm-bcm2708/gcc-linaro-arm-linux-gnueabihf-raspbian-x64/arm-linux-gnueabihf/libc/usr/include/linux/netfilter/xt_TCPMSS.h
        modified:   arm-bcm2708/gcc-linaro-arm-linux-gnueabihf-raspbian-x64/arm-linux-gnueabihf/libc/usr/include/linux/netfilter_ipv4/ipt_ECN.h
        modified:   arm-bcm2708/gcc-linaro-arm-linux-gnueabihf-raspbian-x64/arm-linux-gnueabihf/libc/usr/include/linux/netfilter_ipv4/ipt_TTL.h
        modified:   arm-bcm2708/gcc-linaro-arm-linux-gnueabihf-raspbian-x64/arm-linux-gnueabihf/libc/usr/include/linux/netfilter_ipv6/ip6t_HL.h
        modified:   arm-bcm2708/gcc-linaro-arm-linux-gnueabihf-raspbian/arm-linux-gnueabihf/libc/usr/include/linux/netfilter/xt_CONNMARK.h
        modified:   arm-bcm2708/gcc-linaro-arm-linux-gnueabihf-raspbian/arm-linux-gnueabihf/libc/usr/include/linux/netfilter/xt_DSCP.h
        modified:   arm-bcm2708/gcc-linaro-arm-linux-gnueabihf-raspbian/arm-linux-gnueabihf/libc/usr/include/linux/netfilter/xt_MARK.h
        modified:   arm-bcm2708/gcc-linaro-arm-linux-gnueabihf-raspbian/arm-linux-gnueabihf/libc/usr/include/linux/netfilter/xt_RATEEST.h
        modified:   arm-bcm2708/gcc-linaro-arm-linux-gnueabihf-raspbian/arm-linux-gnueabihf/libc/usr/include/linux/netfilter/xt_TCPMSS.h
        modified:   arm-bcm2708/gcc-linaro-arm-linux-gnueabihf-raspbian/arm-linux-gnueabihf/libc/usr/include/linux/netfilter_ipv4/ipt_ECN.h
        modified:   arm-bcm2708/gcc-linaro-arm-linux-gnueabihf-raspbian/arm-linux-gnueabihf/libc/usr/include/linux/netfilter_ipv4/ipt_TTL.h
        modified:   arm-bcm2708/gcc-linaro-arm-linux-gnueabihf-raspbian/arm-linux-gnueabihf/libc/usr/include/linux/netfilter_ipv6/ip6t_HL.h
```

</details>

Looking at the linux kernel source, there are no such duplicate files (compare https://github.com/torvalds/linux/tree/v4.9/net/netfilter).

When there is no good reason to keep them, I suggest to delete the upper-case files (e.g. `arm-bcm2708/arm-bcm2708-linux-gnueabi/arm-bcm2708-linux-gnueabi/sysroot/usr/include/linux/netfilter/xt_CONNMARK.h`).

popcornmix, 2020-10-10:
These will come from the uapi directory. e.g. https://github.com/torvalds/linux/blob/master/include/uapi/linux/netfilter/xt_CONNMARK.h

jdoubleu, 2020-10-10:
Okay thanks, didn't notice.

However, they just include the lower-case version anyway.