raspberrypi/tools #120: GCC 4.9 needed to compile lastest Raspberry Pi Kernel

upsampled, 2021-02-11:
I was cross-compiling the latest raspbeery pi kernel in the 5.10.y branch and got the following error:

```
cc1: error: -Werror=date-time: no option -Wdate-time
scripts/Makefile.build:279: recipe for target 'scripts/mod/empty.o' failed
make[1]: *** [scripts/mod/empty.o] Error 1
Makefile:1205: recipe for target 'prepare0' failed
```

Looking at a similar issue  by @krichter722 in https://github.com/abperiasamy/rtl8812AU_8821AU_linux/issues/44#issuecomment-100047487, it looks like gcc 4.9 is needed, while gcc 4.8 is provided by this repository:

```
$ arm-linux-gnueabihf-gcc --version
arm-linux-gnueabihf-gcc (crosstool-NG linaro-1.13.1+bzr2650 - Linaro GCC 2014.03) 4.8.3 20140303 (prerelease)
```

dlech, 2021-02-11:
For compiling the kernel, just grab a toolchain from https://developer.arm.com/tools-and-software/open-source-software/developer-tools/gnu-toolchain/gnu-a/downloads.

popcornmix, 2021-02-11:
There are a number of toolchains in this repo and https://github.com/raspberrypi/tools/tree/master/arm-bcm2708/arm-rpi-4.9.3-linux-gnueabihf is 4.9.

But use of these are deprecated there are many alternate sources of arm cross compilers that are frequently updated and will work fine. e.g. on Ubuntu
```
sudo apt install gcc-arm-linux-gnueabihf g++-arm-linux-gnueabihf
```
will be fine.

closed 2021-02-11 by pelwell