raspberrypi/tools #126: E: Package 'gcc-arm-linux-gnueabi' has no installation candidate

capsulecorplab, 2021-12-08:
`$ sudo apt-get install gcc-arm-linux-gnueabi` (as per README) returns an error

```
Reading package lists... Done
Building dependency tree       
Reading state information... Done
Package gcc-arm-linux-gnueabi is not available, but is referred to by another package.
This may mean that the package is missing, has been obsoleted, or
is only available from another source

E: Package 'gcc-arm-linux-gnueabi' has no installation candidate
```

pelwell, 2021-12-09:
It works for me on Bullseye, and it should still work on Buster. Did you `sudo apt update` first? Which OS is installed (`cat /etc/os-release`)?

By the way, `apt-get` is just `apt` now, i.e. `sudo apt install gcc-arm-linux-gnueabi`.

capsulecorplab, 2021-12-14:
> By the way, `apt-get` is just `apt` now, i.e. `sudo apt install gcc-arm-linux-gnueabi`.

I know, I'm just quoting the documentation

capsulecorplab, 2021-12-14:
I ran `sudo apt update && sudo apt upgrade -y` before attempting a `sudo apt install gcc-arm-linux-gnueabi`, but no luck. Attempted on Buster
```
PRETTY_NAME="Raspbian GNU/Linux 10 (buster)"
NAME="Raspbian GNU/Linux"
VERSION_ID="10"
VERSION="10 (buster)"
VERSION_CODENAME=buster
ID=raspbian
ID_LIKE=debian
HOME_URL="http://www.raspbian.org/"
SUPPORT_URL="http://www.raspbian.org/RaspbianForums"
BUG_REPORT_URL="http://www.raspbian.org/RaspbianBugs"
```

pelwell, 2021-12-15:
Which README are you following? I think what you actually want is `gcc-arm-linux-gnueabihf` (note the `hf` for `hard float`), and you'll probably find that it's already installed as `gcc`.

capsulecorplab, 2021-12-15:
Sorry, I must've misread the [readme](https://github.com/raspberrypi/tools/blob/master/README.md) and typed `gcc-arm-linux-gnueabi` instead of `gcc-arm-linux-gnueabihf` :facepalm: but you're right; it looks like `gcc-arm-linux-gnueabihf` is already installed

closed 2021-12-15 by capsulecorplab