raspberrypi/tools #133: Compiling issue

FlyBoyA, 2024-03-13:
Hello Guys,
When I try to compile the kernel & run the command "**make -j4 zImage**" I get the following error.

`naqvi@raspberrypi:~/rpi-kernel/linux $ make -j4 zImage 
./scripts/gcc-version.sh: line 26: /home/naqvi/rpi-kernel/tools/arm-bcm2708/gcc-linaro-arm-linux-gnueabihf-raspbian/bin/arm-linux-gnueabihf-gcc: cannot execute binary file: Exec format error
./scripts/gcc-version.sh: line 27: /home/naqvi/rpi-kernel/tools/arm-bcm2708/gcc-linaro-arm-linux-gnueabihf-raspbian/bin/arm-linux-gnueabihf-gcc: cannot execute binary file: Exec format error
/home/naqvi/rpi-kernel/tools/arm-bcm2708/gcc-linaro-arm-linux-gnueabihf-raspbian/bin/arm-linux-gnueabihf-gcc: 1: Syntax error: "(" unexpected
/home/naqvi/rpi-kernel/tools/arm-bcm2708/gcc-linaro-arm-linux-gnueabihf-raspbian/bin/arm-linux-gnueabihf-gcc: 1: Syntax error: "(" unexpected
/home/naqvi/rpi-kernel/tools/arm-bcm2708/gcc-linaro-arm-linux-gnueabihf-raspbian/bin/arm-linux-gnueabihf-gcc: 1: Syntax error: "(" unexpected
  CHK     include/config/kernel.release
/home/naqvi/rpi-kernel/tools/arm-bcm2708/gcc-linaro-arm-linux-gnueabihf-raspbian/bin/arm-linux-gnueabihf-gcc: 1: Syntax error: "(" unexpected
Cannot use CONFIG_CC_STACKPROTECTOR_REGULAR: -fstack-protector not supported by compiler
/home/naqvi/rpi-kernel/tools/arm-bcm2708/gcc-linaro-arm-linux-gnueabihf-raspbian/bin/arm-linux-gnueabihf-gcc: 1: Syntax error: "(" unexpected
make: *** [Makefile:1115: prepare-compiler-check] Error 1
make: *** Waiting for unfinished jobs....
`

Do you any suggestions?

pelwell, 2024-03-13:
That's an old compiler. Please follow our documented build instructions: https://www.raspberrypi.com/documentation/computers/linux_kernel.html#building

FlyBoyA, 2024-03-15:
I run the command to check 
`make kernelversion`
`4.14.36`

`git branch`
`rpi-6.6.y`

pelwell, 2024-03-15:
Did you read my previous answer?
>  Try "git checkout origin/rpi-4.14.y"

FlyBoyA, 2024-03-16:
Yes I did that and followed to patch the kernel again and the after trying the command below I get the error
`mkae j4 zImage`

`naqvi@raspberrypi:~/rpi-kernel/linuxall $ make -j4 zImage
./scripts/gcc-version.sh: line 26: /home/naqvi/rpi-kernel/tools/arm-bcm2708/gcc-linaro-arm-linux-gnueabihf-raspbian/bin/arm-linux-gnueabihf-gcc: cannot execute binary file: Exec format error
./scripts/gcc-version.sh: line 27: /home/naqvi/rpi-kernel/tools/arm-bcm2708/gcc-linaro-arm-linux-gnueabihf-raspbian/bin/arm-linux-gnueabihf-gcc: cannot execute binary file: Exec format error
/home/naqvi/rpi-kernel/tools/arm-bcm2708/gcc-linaro-arm-linux-gnueabihf-raspbian/bin/arm-linux-gnueabihf-gcc: 1: Syntax error: "(" unexpected
/home/naqvi/rpi-kernel/tools/arm-bcm2708/gcc-linaro-arm-linux-gnueabihf-raspbian/bin/arm-linux-gnueabihf-gcc: 1: Syntax error: "(" unexpected
scripts/kconfig/conf  --silentoldconfig Kconfig
#
# configuration written to .config
#
./scripts/gcc-version.sh: line 26: /home/naqvi/rpi-kernel/tools/arm-bcm2708/gcc-linaro-arm-linux-gnueabihf-raspbian/bin/arm-linux-gnueabihf-gcc: cannot execute binary file: Exec format error
./scripts/gcc-version.sh: line 27: /home/naqvi/rpi-kernel/tools/arm-bcm2708/gcc-linaro-arm-linux-gnueabihf-raspbian/bin/arm-linux-gnueabihf-gcc: cannot execute binary file: Exec format error
/home/naqvi/rpi-kernel/tools/arm-bcm2708/gcc-linaro-arm-linux-gnueabihf-raspbian/bin/arm-linux-gnueabihf-gcc: 1: Syntax error: "(" unexpected
/home/naqvi/rpi-kernel/tools/arm-bcm2708/gcc-linaro-arm-linux-gnueabihf-raspbian/bin/arm-linux-gnueabihf-gcc: 1: Syntax error: "(" unexpected
/home/naqvi/rpi-kernel/tools/arm-bcm2708/gcc-linaro-arm-linux-gnueabihf-raspbian/bin/arm-linux-gnueabihf-gcc: 1: Syntax error: "(" unexpected
  CHK     include/config/kernel.release
/home/naqvi/rpi-kernel/tools/arm-bcm2708/gcc-linaro-arm-linux-gnueabihf-raspbian/bin/arm-linux-gnueabihf-gcc: 1: Syntax error: "(" unexpected
  WRAP    arch/arm/include/generated/uapi/asm/bitsperlong.h
/home/naqvi/rpi-kernel/tools/arm-bcm2708/gcc-linaro-arm-linux-gnueabihf-raspbian/bin/arm-linux-gnueabihf-gcc: 1: Syntax error: "(" unexpected
  WRAP    arch/arm/include/generated/uapi/asm/errno.h
  WRAP    arch/arm/include/generated/uapi/asm/ipcbuf.h
  WRAP    arch/arm/include/generated/uapi/asm/ioctl.h
  WRAP    arch/arm/include/generated/uapi/asm/msgbuf.h
  WRAP    arch/arm/include/generated/uapi/asm/param.h
  WRAP    arch/arm/include/generated/uapi/asm/poll.h
  WRAP    arch/arm/include/generated/uapi/asm/resource.h
  WRAP    arch/arm/include/generated/uapi/asm/sembuf.h
  WRAP    arch/arm/include/generated/uapi/asm/shmbuf.h
  WRAP    arch/arm/include/generated/uapi/asm/siginfo.h
  WRAP    arch/arm/include/generated/uapi/asm/socket.h
  CHK     include/generated/uapi/linux/version.h
  UPD     include/generated/uapi/linux/version.h
  WRAP    arch/arm/include/generated/uapi/asm/sockios.h
  WRAP    arch/arm/include/generated/uapi/asm/termbits.h
  WRAP    arch/arm/include/generated/uapi/asm/termios.h
/home/naqvi/rpi-kernel/tools/arm-bcm2708/gcc-linaro-arm-linux-gnueabihf-raspbian/bin/arm-linux-gnueabihf-gcc: 1: Syntax error: "(" unexpected
  HOSTCC  scripts/basic/bin2c
  UPD     include/config/kernel.release
/home/naqvi/rpi-kernel/tools/arm-bcm2708/gcc-linaro-arm-linux-gnueabihf-raspbian/bin/arm-linux-gnueabihf-gcc: 1: Syntax error: "(" unexpected
  WRAP    arch/arm/include/generated/asm/clkdev.h
  WRAP    arch/arm/include/generated/asm/early_ioremap.h
  WRAP    arch/arm/include/generated/asm/current.h
/home/naqvi/rpi-kernel/tools/arm-bcm2708/gcc-linaro-arm-linux-gnueabihf-raspbian/bin/arm-linux-gnueabihf-gcc: 1: Syntax error: "(" unexpected
  WRAP    arch/arm/include/generated/asm/emergency-restart.h
  WRAP    arch/arm/include/generated/asm/exec.h
  WRAP    arch/arm/include/generated/asm/irq_regs.h
  WRAP    arch/arm/include/generated/asm/extable.h
  WRAP    arch/arm/include/generated/asm/kdebug.h
  CHK     include/generated/utsrelease.h
  WRAP    arch/arm/include/generated/asm/local.h
  WRAP    arch/arm/include/generated/asm/local64.h
  WRAP    arch/arm/include/generated/asm/mm-arch-hooks.h
  WRAP    arch/arm/include/generated/asm/msi.h
  WRAP    arch/arm/include/generated/asm/parport.h
  WRAP    arch/arm/include/generated/asm/preempt.h
  WRAP    arch/arm/include/generated/asm/seccomp.h
  WRAP    arch/arm/include/generated/asm/rwsem.h
  UPD     include/generated/utsrelease.h
  WRAP    arch/arm/include/generated/asm/segment.h
  WRAP    arch/arm/include/generated/asm/serial.h
  WRAP    arch/arm/include/generated/asm/sizes.h
  WRAP    arch/arm/include/generated/asm/simd.h
  WRAP    arch/arm/include/generated/asm/timex.h
  WRAP    arch/arm/include/generated/asm/trace_clock.h
/home/naqvi/rpi-kernel/tools/arm-bcm2708/gcc-linaro-arm-linux-gnueabihf-raspbian/bin/arm-linux-gnueabihf-gcc: 1: Syntax error: "(" unexpected
/home/naqvi/rpi-kernel/tools/arm-bcm2708/gcc-linaro-arm-linux-gnueabihf-raspbian/bin/arm-linux-gnueabihf-gcc: 1: Syntax error: "(" unexpected
/home/naqvi/rpi-kernel/tools/arm-bcm2708/gcc-linaro-arm-linux-gnueabihf-raspbian/bin/arm-linux-gnueabihf-gcc: 1: Syntax error: "(" unexpected
  SYSNR   arch/arm/include/generated/asm/unistd-nr.h
[... 18 lines cut ...]
  HOSTCC  scripts/dtc/livetree.o
  HOSTCC  scripts/dtc/treesource.o
  HOSTCC  scripts/dtc/srcpos.o
/home/naqvi/rpi-kernel/tools/arm-bcm2708/gcc-linaro-arm-linux-gnueabihf-raspbian/bin/arm-linux-gnueabihf-gcc: 1: Syntax error: "(" unexpected
  CC      kernel/bounds.s
[... cut ...]
  HOSTCC  scripts/dtc/dtc-lexer.lex.o
  HOSTCC  scripts/dtc/dtc-parser.tab.o
  HOSTLD  scripts/dtc/dtc
/usr/bin/ld: scripts/dtc/dtc-parser.tab.o:(.bss+0x8): multiple definition of `yylloc'; scripts/dtc/dtc-lexer.lex.o:(.bss+0x1c): first defined here
collect2: error: ld returned 1 exit status
make[2]: *** [scripts/Makefile.host:111: scripts/dtc/dtc] Error 1
make[1]: *** [scripts/Makefile.build:585: scripts/dtc] Error 2
make: *** [Makefile:572: scripts] Error 2
naqvi@raspberrypi:~/rpi-kernel/linuxall $ 
`

FlyBoyA, 2024-03-21:
Do you have any update on this ?

pelwell, 2024-03-21:
It's clear that you are doing something (possibly several things) wrong, in which case this is a request for help - not what this forum is for.

What's the output of `git log --oneline | head`?

FlyBoyA, 2024-03-21:
I understand, I'm not good at it so maybe there could be issue.
that's what I get.
I you could help me and guide me I'll be thankful.
```
naqvi@raspberrypi:~/rpi-kernel/linuxall $ git log --oneline | head
7688b39276ff Backport BCM270X_DT: Rename Pi Zero W DT files
caa653325d58 bcm2835-mmc: Follow renaming of SDHCI_ACMD12_ERR
7e8b161c4d97 Merge tag 'v4.14.114' into rpi-4.14.y
fa5941f45d7e Linux 4.14.114
62c1af5fb31a kernel/sysctl.c: fix out-of-bounds access when setting file-max
599c8a408f87 Revert "locking/lockdep: Add debug_locks check in __lock_downgrade()"
d6b11409047f i2c-hid: properly terminate i2c_hid_dmi_desc_override_table[] array
4024e3bde13e xfs: hold xfs_buf locked between shortform->leaf conversion and the addition of an attribute
2411a27e7475 xfs: add the ability to join a held buffer to a defer_ops
4fa17fc730ba iomap: report collisions between directio and buffered writes to userspace

```