raspberrypi/tools #135: How can I get up to date cross-compiler on x86 platform for Pi3B+ bookworm?

lida2003, 2025-01-01:
How can I get corresponding cross-compiler for 2024-11-25 bookworm-aarch64 ?

```bash
daniel@raspberrypi:~ $ uname -a
Linux raspberrypi 6.6.62+rpt-rpi-v8 #1 SMP PREEMPT Debian 1:6.6.62-1+rpt1 (2024-11-25) aarch64 GNU/Linux
daniel@raspberrypi:~ $ lsb_release -a
No LSB modules are available.
Distributor ID: Debian
Description:    Debian GNU/Linux 12 (bookworm)
Release:        12
Codename:       bookworm
```

6by9, 2025-01-01:
Documentation at https://www.raspberrypi.com/documentation/computers/linux_kernel.html#cross-compile-the-kernel

Under Ubuntu use `sudo apt install crossbuild-essential-arm64` if you're wanting to build a 64bit kernel (based on your 6.6.62+rpt-rpi-v8 kernel version). Other distributions will have similar packages.

lida2003, 2025-01-01:
Is there any zip file for download, which can be simply used on x86 linux system?

Or it's the only way to get crossbuild tool form apt repo?

pelwell, 2025-01-01:
That's a question for your OS community.

closed 2025-01-01 by pelwell